UncleRus/esp-idf-lib #224: Driver request for Sensirion SFA3x sensor ( I2C )

PopherTheG, 2021-07-09:
I am requesting a driver for `SFA3x`.

### Device description

The device is a formaldehyde sensor and it can be controlled through UART or I2C. I think I2C would be a better solution so all Sensirion sensors can be used on one bus. We can use this sensor in tandem to other environmental sensors to better evaluate the air quality.

### Device information

https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/14_Formaldehyde_Sensors/Datasheets/Sensirion_Formaldehyde_Sensors_SFA30_Datasheet.pdf

https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/14_Formaldehyde_Sensors/Application_Note/Sensirion_Formaldehyde_Sensors_AN_SFA30_I2C-Interface_Preliminary.pdf

The product information and the data sheet can be found at:

http://example.org/foohttps://www.sensirion.com/en/download-center/formaldehyde-sensors/formaldehyde-sensor-module/

### Purchasing the product

The device can be purchased at:

- [ ] Aliexpress
- [ ] Amazon
- [ ] ebay
- [x] Digi-Key

The URLs where you can buy the device are:

- https://www.digikey.com/en/products/detail/sensirion-ag/SEK-SFA30/13577460?s=N4IgTCBcDaIM4DMCGBmADCAugXyA

### Other implementation

There is a driver code present in their GitHub repository: https://github.com/Sensirion/embedded-sfa3x

UncleRus, 2021-07-10:
@PopherTheG Could you test the driver? #225

PopherTheG, 2021-07-10:
@UncleRus  Yes I can. I will test it when I get home, since my device is set up there.  Around 5 hours from now I will have tested it.

PopherTheG, 2021-07-10:
I tried running the example and I seem to get this error logged on the IDF terminal.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6896
load:0x40078000,len:14292
ho 0 tail 12 room 4
load:0x40080400,len:3688
0x40080400: _init at ??:?

entry 0x40080678
I (29) boot: ESP-IDF v4.3 2nd stage bootloader
I (29) boot: compile time 22:49:51
I (29) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=085b4h ( 34228) map
I (124) esp_image: segment 1: paddr=000185dc vaddr=3ffb0000 size=02908h ( 10504) load
I (128) esp_image: segment 2: paddr=0001aeec vaddr=40080000 size=0512ch ( 20780) load
I (139) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1663ch ( 91708) map
I (174) esp_image: segment 4: paddr=00036664 vaddr=4008512c size=06ff8h ( 28664) load
I (186) esp_image: segment 5: paddr=0003d664 vaddr=50000000 size=00010h (    16) load
I (193) boot: Loaded app from partition at offset 0x10000
I (193) boot: Disabling RNG early entropy source...
I (206) cpu_start: Pro cpu up.
I (206) cpu_start: Starting app cpu, entry point is 0x40081104
0x40081104: call_start_cpu1 at C:/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (221) cpu_start: Pro cpu start user code
I (221) cpu_start: cpu freq: 160000000
I (221) cpu_start: Application information:
I (225) cpu_start: Project name:     example-sfa3x
I (231) cpu_start: App version:      0.8.0-71-g186799d
I (236) cpu_start: Compile time:     Jul 10 2021 22:49:13
I (242) cpu_start: ELF file SHA256:  e631ad557a98e9c9...
I (248) cpu_start: ESP-IDF:          v4.3
I (253) heap_init: Initializing. RAM available for dynamic allocation:
I (260) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (266) heap_init: At 3FFB3168 len 0002CE98 (179 KiB): DRAM
I (273) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (279) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (285) heap_init: At 4008C124 len 00013EDC (79 KiB): IRAM
I (293) spi_flash: detected chip: generic
I (296) spi_flash: flash io: dio
W (300) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (314) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (20) i2cdev: Could not write to device [0x5d at 0]: 263
ESP_ERROR_CHECK failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x400841f0
0x400841f0: _esp_error_check_failed at C:/esp/esp-idf/components/esp_common/src/esp_err.c:41

file: "../main/main.c" line 27
func: task
expression: sfa3x_reset(&dev)

abort() was called at PC 0x400841f3 on core 1
0x400841f3: _esp_error_check_failed at C:/esp/esp-idf/components/esp_common/src/esp_err.c:42

Backtrace:0x40085cdf:0x3ffb8b70 0x40086405:0x3ffb8b90 0x4008b17a:0x3ffb8bb0 0x400841f3:0x3ffb8c20 0x400d51e1:0x3ffb8c40 0x40088d45:0x3ffb8cd0
0x40085cdf: panic_abort at C:/esp/esp-idf/components/esp_system/panic.c:356

0x40086405: esp_system_abort at C:/esp/esp-idf/components/esp_system/system_api.c:112

0x4008b17a: abort at C:/esp/esp-idf/components/newlib/abort.c:46

0x400841f3: _esp_error_check_failed at C:/esp/esp-idf/components/esp_common/src/esp_err.c:42

0x400d51e1: task at d:\esp-idf-lib\examples\sfa3x\build/../main/main.c:27 (discriminator 1)

0x40088d45: vPortTaskWrapper at C:/esp/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: e631ad557a98e9c9

Rebooting...
ets Jun  8 2016 00:22:57

```

The example sfa3x has this behavior that when the code starts it just reboots the ESP32. I will try to fix and find where it is crashing.

UncleRus, 2021-07-10:
Please build and run i2c scanner: https://github.com/UncleRus/esp-idf-i2cscan and show console output.

UncleRus, 2021-07-10:
It crashing because of `ESP_ERROR_CHECK` macro.
The `sfa30_reset()` function tries to write command to I2C device with address 0x5d, but fails.
So we need to scan the entire I2C bus and understand which devices with which addresses are available on it.

PopherTheG, 2021-07-10:
I am getting this output:

[image]

I saw something weird when I tried removing the ESP_ERROR_CHECK it shows some values. But I'm not sure if it is trash values. 

[image]

```
I (0) cpu_start: App cpu up.
I (220) cpu_start: Pro cpu start user code
I (221) cpu_start: cpu freq: 160000000
I (221) cpu_start: Application information:
I (225) cpu_start: Project name:     example-sfa3x
I (231) cpu_start: App version:      0.8.0-71-g186799d
I (236) cpu_start: Compile time:     Jul 10 2021 22:49:13
I (242) cpu_start: ELF file SHA256:  11cdeb0086a47210...
I (248) cpu_start: ESP-IDF:          v4.3
I (253) heap_init: Initializing. RAM available for dynamic allocation:
I (260) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (266) heap_init: At 3FFB3168 len 0002CE98 (179 KiB): DRAM
I (273) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (279) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (285) heap_init: At 4008C124 len 00013EDC (79 KiB): IRAM
I (293) spi_flash: detected chip: generic
I (296) spi_flash: flash io: dio
W (300) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (314) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (20) i2cdev: Could not write to device [0x5d at 0]: 263
I (20) sfa3x-example: Sensor reset
E (30) sfa3x: Invalid CRC 0xac, expected 0xff
I (30) sfa3x-example: Sensor marking:
I (30) sfa3x-example: Continuous measurement started
I (550) sfa3x-example: Formaldehyde: 7.80 bpm
I (550) sfa3x-example: Temperature: 29.00 °C
I (550) sfa3x-example: Humidity: 78.25 %
I (1060) sfa3x-example: Formaldehyde: 5.80 bpm
I (1060) sfa3x-example: Temperature: 29.02 °C
I (1060) sfa3x-example: Humidity: 78.09 %
I (1570) sfa3x-example: Formaldehyde: 5.40 bpm
I (1570) sfa3x-example: Temperature: 29.03 °C
I (1570) sfa3x-example: Humidity: 78.26 %
I (2080) sfa3x-example: Formaldehyde: 5.40 bpm
I (2080) sfa3x-example: Temperature: 29.02 °C
I (2080) sfa3x-example: Humidity: 78.36 %
I (2590) sfa3x-example: Formaldehyde: 5.60 bpm
I (2590) sfa3x-example: Temperature: 29.06 °C
I (2590) sfa3x-example: Humidity: 78.39 %
I (3100) sfa3x-example: Formaldehyde: 5.60 bpm
I (3100) sfa3x-example: Temperature: 29.07 °C
I (3100) sfa3x-example: Humidity: 78.43 %
I (3610) sfa3x-example: Formaldehyde: 5.60 bpm
I (3610) sfa3x-example: Temperature: 29.02 °C
I (3610) sfa3x-example: Humidity: 78.40 %
```

When the ESP_ERROR_CHECK is there it crashes.

UncleRus, 2021-07-10:
So, the reset function doesn't work (timeout during writing, the device does not responding by ACK) and we cannot read the marking (CRC error), but the start of measurements and the reading of values work.

What can we try:
- Add `vTaskDelay(pdMS_TO_TICKS(5000))` before `sfa3x_reset()`: device still may be not initialized at the moment we trying to reset it.
- In configuration (`idf.py menuconfig`) set default log level to verbose to show sent/received bytes to device. The driver swaps bytes before calculating the CRC when reading the marking and this may be wrong.

UncleRus, 2021-07-10:
> When the ESP_ERROR_CHECK is there it crashes.

Of course, this is the main purpose of `ESP_ERROR_CHECK()` macro.

UncleRus, 2021-07-10:
> The driver swaps bytes before calculating the CRC when reading the marking and this may be wrong.

No, it's not...

@PopherTheG could you set default log level to verbose?

`idf.py menuconfig -> Component config -> Log output -> Default log verbosity -> Verbose`
or manually change in sdkconfig file of example project
```
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
...
# CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE is not set
```
to 
```
# CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
...
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=y
```

Please post all console output here after that.

PopherTheG, 2021-07-11:
Here is the output log after adding the vTaskDelay before the reset command:

[image]

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6896
load:0x40078000,len:14292
ho 0 tail 12 room 4
load:0x40080400,len:3688
0x40080400: _init at ??:?

entry 0x40080678
I (29) boot: ESP-IDF v4.3 2nd stage bootloader
I (29) boot: compile time 19:31:18
I (29) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=08d8ch ( 36236) map
I (124) esp_image: segment 1: paddr=00018db4 vaddr=3ffb0000 size=029dch ( 10716) load
I (129) esp_image: segment 2: paddr=0001b798 vaddr=40080000 size=04880h ( 18560) load
I (138) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=16c08h ( 93192) map
I (174) esp_image: segment 4: paddr=00036c30 vaddr=40084880 size=0795ch ( 31068) load
I (187) esp_image: segment 5: paddr=0003e594 vaddr=50000000 size=00010h (    16) load
I (194) boot: Loaded app from partition at offset 0x10000
I (195) boot: Disabling RNG early entropy source...
I (207) cpu_start: Pro cpu up.
I (208) cpu_start: Starting app cpu, entry point is 0x40081124
0x40081124: call_start_cpu1 at C:/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
D (215) clk: RTC_SLOW_CLK calibration value: 3430797
I (227) cpu_start: Pro cpu start user code
I (227) cpu_start: cpu freq: 160000000
I (227) cpu_start: Application information:
I (231) cpu_start: Project name:     example-sfa3x
I (237) cpu_start: App version:      0.8.0-71-g186799d-dirty
I (243) cpu_start: Compile time:     Jul 11 2021 19:30:51
I (249) cpu_start: ELF file SHA256:  b35696b3d9d4aba9...
I (255) cpu_start: ESP-IDF:          v4.3
V (260) memory_layout: reserved range is 0x3f408d74 - 0x3f408dac
D (266) memory_layout: Checking 7 reserved memory ranges:
D (271) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (278) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb3240
D (284) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (291) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (297) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (304) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at C:/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1730

D (310) memory_layout: Reserved memory range 0x40080000 - 0x4008c1dc
0x40080000: _WindowOverflow4 at C:/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1730

D (316) memory_layout: Building list of available memory regions:
V (323) memory_layout: Examining memory region 0x3ffae000 - 0x3ffb0000
V (329) memory_layout: Start of region 0x3ffae000 - 0x3ffb0000 overlaps reserved 0x3ffae000 - 0x3ffae6e0
D (339) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
[... 243 lines cut ...]
0x4008b1ce: abort at C:/esp/esp-idf/components/newlib/abort.c:46

0x40084247: _esp_error_check_failed at C:/esp/esp-idf/components/esp_common/src/esp_err.c:42

0x400d546f: task at d:\esp-idf-lib\examples\sfa3x\build/../main/main.c:28 (discriminator 1)

0x40088d99: vPortTaskWrapper at C:/esp/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: b35696b3d9d4aba9

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6896
load:0x40078000,len:14292
ho 0 tail 12 room 4
load:0x40080400,len:3688
0x40080400: _init at ??:?

entry 0x40080678
```

UncleRus, 2021-07-11:
Please comment out the line 27 in example:
```C
...
//ESP_ERROR_CHECK(sfa3x_reset(&dev));
```
and run example again.

rehpopmegnoz, 2021-07-12:
Hi @UncleRus I will test it again later when I get home. Ill be home around 7PM July 12 2021 GMT+8.

PopherTheG, 2021-07-12:
Hi @UncleRus I will test it again later when I get home. Ill be home around 7PM July 12 2021 GMT+8.

PopherTheG, 2021-07-12:
Here is the output log when `sfa3x_reset(&dev)` is commented out.

```
I (29) boot: ESP-IDF v4.3 2nd stage bootloader
I (29) boot: compile time 19:31:18
I (29) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=08d7ch ( 36220) map
I (125) esp_image: segment 1: paddr=00018da4 vaddr=3ffb0000 size=029dch ( 10716) load
I (129) esp_image: segment 2: paddr=0001b788 vaddr=40080000 size=04890h ( 18576) load
I (138) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=16bc4h ( 93124) map
I (174) esp_image: segment 4: paddr=00036bec vaddr=40084890 size=0794ch ( 31052) load
I (187) esp_image: segment 5: paddr=0003e540 vaddr=50000000 size=00010h (    16) load
I (195) boot: Loaded app from partition at offset 0x10000
I (195) boot: Disabling RNG early entropy source...
I (208) cpu_start: Pro cpu up.
I (208) cpu_start: Starting app cpu, entry point is 0x40081124
0x40081124: call_start_cpu1 at C:/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (193) cpu_start: App cpu up.
D (215) clk: RTC_SLOW_CLK calibration value: 3436083
I (227) cpu_start: Pro cpu start user code
I (227) cpu_start: cpu freq: 160000000
I (227) cpu_start: Application information:
I (232) cpu_start: Project name:     example-sfa3x
I (237) cpu_start: App version:      0.8.0-71-g186799d-dirty
I (243) cpu_start: Compile time:     Jul 11 2021 19:30:51
I (250) cpu_start: ELF file SHA256:  e04677b7eeae6079...
I (256) cpu_start: ESP-IDF:          v4.3
V (260) memory_layout: reserved range is 0x3f408d64 - 0x3f408d9c
D (266) memory_layout: Checking 7 reserved memory ranges:
D (272) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (278) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb3240
D (285) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (291) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (297) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (304) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at C:/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1730

D (310) memory_layout: Reserved memory range 0x40080000 - 0x4008c1dc
0x40080000: _WindowOverflow4 at C:/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1730

D (317) memory_layout: Building list of available memory regions:
V (323) memory_layout: Examining memory region 0x3ffae000 - 0x3ffb0000
V (329) memory_layout: Start of region 0x3ffae000 - 0x3ffb0000 overlaps reserved 0x3ffae000 - 0x3ffae6e0
D (339) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
V (346) memory_layout: Examining memory region 0x3ffb0000 - 0x3ffb8000
V (352) memory_layout: Start of region 0x3ffb0000 - 0x3ffb8000 overlaps reserved 0x3ffb0000 - 0x3ffb3240
D (362) memory_layout: Available memory region 0x3ffb3240 - 0x3ffb8000
V (368) memory_layout: Examining memory region 0x3ffb8000 - 0x3ffc0000
D (375) memory_layout: Available memory region 0x3ffb8000 - 0x3ffc0000
V (382) memory_layout: Examining memory region 0x3ffc0000 - 0x3ffc2000
D (388) memory_layout: Available memory region 0x3ffc0000 - 0x3ffc2000
V (395) memory_layout: Examining memory region 0x3ffc2000 - 0x3ffc4000
D (401) memory_layout: Available memory region 0x3ffc2000 - 0x3ffc4000
V (408) memory_layout: Examining memory region 0x3ffc4000 - 0x3ffc6000
D (415) memory_layout: Available memory region 0x3ffc4000 - 0x3ffc6000
V (421) memory_layout: Examining memory region 0x3ffc6000 - 0x3ffc8000
D (428) memory_layout: Available memory region 0x3ffc6000 - 0x3ffc8000
[... 274 lines cut ...]
    float hcho, temperature, humidity;
    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(500));

        esp_err_t res = sfa3x_read_measurement(&dev, &hcho, &humidity, &temperature);
        if (res != ESP_OK)
        {
            ESP_LOGE(TAG, "Error reading results %d (%s)", res, esp_err_to_name(res));
            continue;
        }

        ESP_LOGI(TAG, "Formaldehyde: %.2f bpm", hcho);
        ESP_LOGI(TAG, "Temperature: %.2f °C", temperature);
        ESP_LOGI(TAG, "Humidity: %.2f %%", humidity);
    }
}

void app_main()
{
    ESP_ERROR_CHECK(i2cdev_init());

    xTaskCreatePinnedToCore(task, TAG, configMINIMAL_STACK_SIZE * 8, NULL, 5, NULL, APP_CPU_NUM);
}
```

PopherTheG, 2021-07-15:
Is it safe to say that I should proceed with just using `ESP_ERROR_CHECK_WITHOUT_ABORT`?

UncleRus, 2021-07-16:
So, we have a problem here: your module does not respond to the "reset" and "read marking" commands. Considering that the datasheet contains a "preliminary" mark, there is most likely some kind of discrepancy between the datasheet and the firmware of your module.

> Is it safe to say that I should proceed with just using ESP_ERROR_CHECK_WITHOUT_ABORT?

I don't understand your question. Considering that your module does not respond correctly to the "reset" and "read marking" commands, just do not call the corresponding functions (`sfa30_reset()` and `sfa3x_get_device_marknig()`) at all.

But I am unable to include this driver in the library in its current state. Thus, I leave this issue open pending help from the owners of this module (possibly with a different firmware version).

PR #225 (2023-08-21): feature: driver + example for SFA30
Fix #224.
@PopherTheG Please test
commit 65d64af0 (2023-08-21): feature: driver + example for SFA30, fixes #224
closed 2023-08-21 by UncleRus