UncleRus/esp-idf-lib #226: sht3x example bug - ESP32-D0WD

xaos11, 2021-07-09:
**Device type**:

- [x] ESP32-D0WD (M5stack Core2)

**Framework version**:

- [x] ESP-IDF (PlatformIO)

**Describe the bug**:

[Sht3x example code](https://github.com/UncleRus/esp-idf-lib/tree/master/examples/sht3x) fails with the following output:

␛[0;31mE (864) i2cdev: Could not write to device [0x44 at 0]: -1␛[0m
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40085290

Through some trial and error I found that commenting out this line gives me the desired output:

`\\ESP_ERROR_CHECK(sht3x_init(&dev));`

SHT3x Sensor: 25.42 °C, 46.78 %
SHT3x Sensor: 25.45 °C, 46.88 %
SHT3x Sensor: 25.45 °C, 46.88 %
SHT3x Sensor: 25.45 °C, 46.84 %
SHT3x Sensor: 25.43 °C, 46.80 %
SHT3x Sensor: 25.46 °C, 46.81 %

I'm not sure what this means or what the risk is, but it fixed it for me.

UncleRus, 2021-07-09:
Can't reproduce this bug.
Please check your wiring and pull-up resistors.

xaos11, 2021-07-09:
The hardware I'm using seems to have the pull-up resistor built in. Maybe it's just a result of using this specific hardware? I would think the wiring should be good as it's connected via the Grove connector.

https://docs.m5stack.com/en/unit/envII

zenz, 2021-07-20:
same situation here. I tried Lolin D32 Lite. TTGo-T7, NodeMCU.

UncleRus, 2022-05-10:
It looks like SHT3x may not ACK on reset command:

https://github.com/UncleRus/esp-idf-lib/blob/49ea72192e81d60d31abf8f89753efaa7368ebba/components/sht3x/sht3x.c#L239

UncleRus, 2022-05-10:
Could you test?

commit 69d4f6fe (2022-05-10): Revert "sht3x: fix #226"

This reverts commit 8738d893884c1c4e88de82c2840228de2bc462a3.
commit b8898487 (2022-05-10): sht3x: fix #226
closed 2022-05-10 by UncleRus