UncleRus/esp-idf-lib #229: About max7219 Driver

nopnop2002, 2021-07-13:
**Device type**:

- [x] ESP32
- [ ] ESP8226

**Framework version**:

- [x] ESP-IDF latest master
- [ ] ESP-IDF 3.2.x
- [ ] ESP-IDF 3.3.x
- [ ] ESP8266 RTOS SDK latest master
- [ ] ESP8266 RTOS SDK 3.2

**Describe the bug**:

Thank you for publishing a useful library.

I bought this 8x8 matrix LED.

[image]

So I tried [this](https://github.com/UncleRus/esp-idf-lib/tree/master/examples/max7219_8x8) sample code.

First I changed this.
```
//#define CASCADE_SIZE 1
#define CASCADE_SIZE 4
```

I don't understand exactly what the mirrored variable means.

```
    max7219_t dev = {
       .cascade_size = CASCADE_SIZE,
       .digits = 0,
       .mirrored = true
    };
```

I added some ESP_LOGI.
```
esp_err_t max7219_set_digit(max7219_t *dev, uint8_t digit, uint8_t val)
{
    ESP_LOGI(TAG, "digit=%d", digit);
    CHECK_ARG(dev);
    if (digit >= dev->digits)
    {
        ESP_LOGE(TAG, "Invalid digit: %d", digit);
        return ESP_ERR_INVALID_ARG;
    }

    if (dev->mirrored)
        digit = dev->digits - digit - 1;

    uint8_t c = digit / ALL_DIGITS;
    uint8_t d = digit % ALL_DIGITS;

    ESP_LOGI(TAG, "Chip %d, digit %d val 0x%02x", c, d, val);

    CHECK(send(dev, c, (REG_DIGIT_0 + ((uint16_t)d << 8)) | val));

    return ESP_OK;
}
```

# Case of mirrored = true
```
---------- draw
I (2374) max7219: digit=0
I (2374) max7219: Chip 3, digit 7 val 0x38
I (2374) max7219: digit=1
I (2374) max7219: Chip 3, digit 6 val 0x38
I (2374) max7219: digit=2
I (2374) max7219: Chip 3, digit 5 val 0x00
I (2384) max7219: digit=3
I (2384) max7219: Chip 3, digit 4 val 0x38
I (2394) max7219: digit=4
I (2394) max7219: Chip 3, digit 3 val 0x38
I (2404) max7219: digit=5
I (2404) max7219: Chip 3, digit 2 val 0x38
I (2404) max7219: digit=6
I (2414) max7219: Chip 3, digit 1 val 0xfe
I (2414) max7219: digit=7
I (2424) max7219: Chip 3, digit 0 val 0x7c
I (2424) max7219: digit=1
I (2424) max7219: Chip 3, digit 6 val 0x38
I (2434) max7219: digit=2
I (2434) max7219: Chip 3, digit 5 val 0x10
I (2444) max7219: digit=3
I (2444) max7219: Chip 3, digit 4 val 0x00
I (2444) max7219: digit=4
I (2454) max7219: Chip 3, digit 3 val 0x10
I (2454) max7219: digit=5
I (2464) max7219: Chip 3, digit 2 val 0x30
I (2464) max7219: digit=6
I (2464) max7219: Chip 3, digit 1 val 0x7e
I (2474) max7219: digit=7
I (2474) max7219: Chip 3, digit 0 val 0xfe
I (2484) max7219: digit=8
I (2484) max7219: Chip 2, digit 7 val 0x7e
I (2484) max7219: digit=2
I (2494) max7219: Chip 3, digit 5 val 0x30
I (2494) max7219: digit=3
I (2504) max7219: Chip 3, digit 4 val 0x10
I (2504) max7219: digit=4
I (2504) max7219: Chip 3, digit 3 val 0x00
I (2514) max7219: digit=5
I (2514) max7219: Chip 3, digit 2 val 0x10
I (2524) max7219: digit=6
I (2524) max7219: Chip 3, digit 1 val 0x18
I (2524) max7219: digit=7
I (2534) max7219: Chip 3, digit 0 val 0xfc
I (2534) max7219: digit=8
I (2544) max7219: Chip 2, digit 7 val 0xfe
I (2544) max7219: digit=9
I (2544) max7219: Chip 2, digit 6 val 0xfc
I (2554) max7219: digit=3
I (2554) max7219: Chip 3, digit 4 val 0x18
I (2564) max7219: digit=4
I (2564) max7219: Chip 3, digit 3 val 0x10
I (2574) max7219: digit=5
I (2574) max7219: Chip 3, digit 2 val 0x00
I (2574) max7219: digit=6
I (2584) max7219: Chip 3, digit 1 val 0x44
I (2584) max7219: digit=7
I (2584) max7219: Chip 3, digit 0 val 0xee
I (2594) max7219: digit=8
I (2594) max7219: Chip 2, digit 7 val 0xfe
I (2604) max7219: digit=9
I (2604) max7219: Chip 2, digit 6 val 0xfe
I (2614) max7219: digit=10
I (2614) max7219: Chip 2, digit 5 val 0x7c
---------- draw
```

# Case of mirrored = false
Is the value of the digit parameter of max7219_set_digit() correct?

```
---------- draw
I (2374) max7219: digit=0
I (2374) max7219: Chip 0, digit 0 val 0x38
I (2374) max7219: digit=1
I (2374) max7219: Chip 0, digit 1 val 0x38
I (2374) max7219: digit=2
I (2374) max7219: Chip 0, digit 2 val 0x00
I (2384) max7219: digit=3
I (2384) max7219: Chip 0, digit 3 val 0x38
I (2394) max7219: digit=4
I (2394) max7219: Chip 0, digit 4 val 0x38
I (2404) max7219: digit=5
I (2404) max7219: Chip 0, digit 5 val 0x38
I (2404) max7219: digit=6
I (2414) max7219: Chip 0, digit 6 val 0xfe
I (2414) max7219: digit=7
I (2424) max7219: Chip 0, digit 7 val 0x7c
I (2424) max7219: digit=1 ---> 8?
I (2424) max7219: Chip 0, digit 1 val 0x38
I (2434) max7219: digit=2 ---> 9?
I (2434) max7219: Chip 0, digit 2 val 0x10
I (2444) max7219: digit=3 ---> 10?
I (2444) max7219: Chip 0, digit 3 val 0x00
I (2444) max7219: digit=4
I (2454) max7219: Chip 0, digit 4 val 0x10
I (2454) max7219: digit=5
I (2464) max7219: Chip 0, digit 5 val 0x30
I (2464) max7219: digit=6
I (2464) max7219: Chip 0, digit 6 val 0x7e
I (2474) max7219: digit=7
I (2474) max7219: Chip 0, digit 7 val 0xfe
I (2484) max7219: digit=8
I (2484) max7219: Chip 1, digit 0 val 0x7e
I (2484) max7219: digit=2
I (2494) max7219: Chip 0, digit 2 val 0x30
I (2494) max7219: digit=3
I (2504) max7219: Chip 0, digit 3 val 0x10
I (2504) max7219: digit=4
I (2504) max7219: Chip 0, digit 4 val 0x00
I (2514) max7219: digit=5
I (2514) max7219: Chip 0, digit 5 val 0x10
I (2524) max7219: digit=6
I (2524) max7219: Chip 0, digit 6 val 0x18
I (2524) max7219: digit=7
I (2534) max7219: Chip 0, digit 7 val 0xfc
I (2534) max7219: digit=8
I (2544) max7219: Chip 1, digit 0 val 0xfe
I (2544) max7219: digit=9
I (2544) max7219: Chip 1, digit 1 val 0xfc
I (2554) max7219: digit=3
I (2554) max7219: Chip 0, digit 3 val 0x18
I (2564) max7219: digit=4
I (2564) max7219: Chip 0, digit 4 val 0x10
I (2574) max7219: digit=5
I (2574) max7219: Chip 0, digit 5 val 0x00
I (2574) max7219: digit=6
I (2584) max7219: Chip 0, digit 6 val 0x44
I (2584) max7219: digit=7
I (2584) max7219: Chip 0, digit 7 val 0xee
I (2594) max7219: digit=8
I (2594) max7219: Chip 1, digit 0 val 0xfe
I (2604) max7219: digit=9
I (2604) max7219: Chip 1, digit 1 val 0xfe
I (2614) max7219: digit=10
I (2614) max7219: Chip 1, digit 2 val 0x7c
---------- draw
```

Thank you.

UncleRus, 2021-07-13:
`digit` parameter in `max7219_set_digit()` is end-to-end number of digit for all devices in the cascade. Digit is 8 LEDs (one eight-segments display or, in your case, one row or column of a matrix display). One MAX7219 supports up to 8 digits and you have 4 devices cascaded. So for you setup `digit` value range is 8*4 = 32.

In some setups last device in the cascade can have less than 8 digits actually connected to LEDs and in this case you can set number of available digits manually in the device descriptor before you call `max7219_init()`. Otherwise, if you leave a zero in the `digits` field in the descriptor it will be automatically calculated and set in `max7219_init()` to a maximal available value (`cascade_size * 8`).

```
...
max7219_t dev = {
   .cascade_size = 4,
   .digits = 0, // calculate number of digits automatically
   .mirrored = true
};
max7219_init_desc(&dev, HOST, PIN_NUM_CS);
max7219_init(&dev);
// now dev.digits = 32
...
for (uint8_t d = 0; d < dev.digits; d ++)
    max7219_set_digit(&dev, d, (d & 1) ? 0xaa : 0x55); // draw something like a checkerboard
...
```
The `mirrored` flag used in cases when order of the digits in the actual device is reversed. You can try both variants and see which works better for you setup.

nopnop2002, 2021-07-13:
Thank you for your reply.

I understand the meaning of [digits] and [mirrored].
 
I changed one line.

It can now be displayed correctly when CASCADE_SIZE = 4.

```
esp_err_t max7219_draw_image_8x8(max7219_t *dev, uint8_t pos, const void *image)
{
    ESP_LOGI(TAG, "max7219_draw_image_8x8 pos=%d", pos);
    CHECK_ARG(dev && image);

    //for (uint8_t i = pos, offs = 0; i < dev->digits && offs < 8; i++, offs++)
    for (uint8_t i = (pos*8), offs = 0; i < dev->digits && offs < 8; i++, offs++)
        max7219_set_digit(dev, i, *((uint8_t *)image + offs));

    return ESP_OK;
}
```

UncleRus, 2021-07-13:
As far as I understand, you are implying that the `pos` argument is the number of chip in the cascade, so you multiply it by 8 in your edit. But the driver assumes that this argument means the *digit*, not the chip. For example, this code draws half of symbol on the first 8x8 matrix and half - on the second.
```C
max7219_draw_image_8x8(&dev, 4, symbol);
```
With your edit, this is not possible.

nopnop2002, 2021-07-13:
This is original code:
[image]

This is my code:
[image]

UncleRus, 2021-07-13:
Ahhh, a bug in my example...
Thank you for reporting!

closed 2021-07-13 by UncleRus

nopnop2002, 2021-07-14:
I fonud one more problem.
I changed it like this:

```
    while (1) {
        printf("---------- draw\n");

        for (uint8_t c = 0; c < CASCADE_SIZE; c ++)
            max7219_draw_image_8x8(&dev, c * 8, (uint8_t *)symbols + c * 8 + offs);

        printf("offs=%d symbols_size=%d\n", offs, symbols_size);
        if (++offs == symbols_size) break;
    }
    vTaskDelete(NULL);
```

Not displayed until the end.

[image]

---

```
    size_t offs = 0;
    while (1) {
        printf("---------- draw\n");

        for (uint8_t c = 0; c < CASCADE_SIZE; c ++)
            max7219_draw_image_8x8(&dev, c * 8, (uint8_t *)symbols + c * 8 + offs);

        printf("offs=%d symbols_size=%d\n", offs, symbols_size);
        if (++offs > symbols_size) break;
        //if (++offs == symbols_size) break;
    }
    vTaskDelete(NULL);
```

displayed until the end.

[image]

---

If mirrored = false, the numbers are reversed.

[image]

UncleRus, 2021-07-16:
I don't think it's a good idea to implement a specific display layout in a driver.

Rotating / transposing images is the application task.

nopnop2002, 2021-07-16:
>Image rotation / transposition is an application task.

I agree.

commit 72936bc7 (2021-07-13): max7219: fix bug with chip number in 8x8 example (fixes #229)