UncleRus/esp-idf-lib #230: Driver request for TCA6416A GPIO expander from Texas Instruments

rehpopmegnoz, 2021-07-13:
I am requesting a driver for `TCA6416A`.

### Device description

TCA6416A is a 16-bit I2C GPIO expander, which can also send an interrupt thru its interrupt pin to signify a microcontroller that one of the input in the GPIO expander changed state.

### Device information

The product information and the data sheet can be found at:

https://www.ti.com/product/TCA6416A?utm_source=google&utm_medium=cpc&utm_campaign=asc-null-null-GPN_EN-cpc-pf-google-sa&utm_content=TCA6416A&ds_k=TCA6416A&DCM=yes&gclsrc=ds&gclsrc=ds#tech-docs

https://www.ti.com/lit/ds/symlink/tca6416a.pdf?ts=1626141249235&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FTCA6416A%253Futm_source%253Dgoogle%2526utm_medium%253Dcpc%2526utm_campaign%253Dasc-int-null-prodfolderdynamic-cpc-pf-google-kr%2526utm_content%253Dprodfolddynamic%2526ds_k%253DDYNAMIC%2BSEARCH%2BADS%2526DCM%253Dyes%2526gclid%253DEAIaIQobChMInq_Ijvje8QIVVK2WCh02CARcEAAYASAAEgLl7PD_BwE%2526gclsrc%253Daw.ds

### Purchasing the product

The device can be purchased at:

- [ ] Aliexpress
- [ ] Amazon
- [ ] ebay
- [x] Digi-Key

The URLs where you can buy the device are:

- https://www.digikey.com/en/product-highlight/t/texas-instruments/tca6416a-expander

### Other implementation

There is driver code posted here in the 'Software Development' tab: https://www.ti.com/product/TCA6416A?utm_source=google&utm_medium=cpc&utm_campaign=asc-null-null-GPN_EN-cpc-pf-google-sa&utm_content=TCA6416A&ds_k=TCA6416A&DCM=yes&gclsrc=ds&gclsrc=ds#design-development##software-development

PopherTheG, 2021-07-14:
Hi! I used the TCA95x5 for this and it seems to work normally. I will just write an issue if ever something different comes up.

UncleRus, 2021-07-15:
TCA6416A is already supported by `tca95x5` driver.

closed 2021-07-15 by UncleRus