UncleRus/esp-idf-lib #231: Requesting a driver for `MAX7219`.

nopnop2002, 2021-07-14:
I am requesting a driver for `MAX7219`.

Add one parameter to max7219_draw_image_8x8.

```
esp_err_t max7219_draw_image_8x8(max7219_t *dev, uint8_t pos, const void *image, bool flag)
```
---

If flag = true, immediately issue an SPI to the device.

```
        for (uint8_t c = 0; c < CASCADE_SIZE; c ++)
            max7219_draw_image_8x8(&dev, c * 8, (uint8_t *)symbols + c * 8 + offs, true);
```

Such data is output to SPI:
```
[0x01 0xff]
[0x01 0xff 0x00 0x00]
[0x01 0xff 0x00 0x00 0x00 0x00]
[0x01 0xff 0x00 0x00 0x00 0x00 0x00 0x00]
```

The bottom row of the four Matrix LEDs turns ON.

--- 

If flag = false, data is saved in the [INTERNAL buffer] and SPI is not issued to the device.
[INTERNAL buffer] is uint8_t buffer[8][8].   
When max7219_flash is executed, SPI is issued to the device based on the data in the [INTERNAL buffer].

```
        for (uint8_t c = 0; c < CASCADE_SIZE; c ++)
            max7219_draw_image_8x8(&dev, c * 8, (uint8_t *)symbols + c * 8 + offs, false);
        max7219_flash(&dev);
```

Such data is output to SPI:
```
[0x01 0xff 0x01 0xff 0x01 0xff 0x01 0xff]
```

The bottom row of the four Matrix LEDs turns ON.

It can be displayed at high speed.

Thank you.

UncleRus, 2021-07-16:
Good idea, but implementing it will break compatibility.
I will think about it.

nopnop2002, 2021-07-16:
How about hiding the flag variable inside the structure like this?

```
typedef struct
{
    spi_device_interface_config_t spi_cfg;
    spi_device_handle_t spi_dev;
    uint8_t digits;              //!< Accessible digits in 7seg. Up to cascade_size * 8
    uint8_t cascade_size;        //!< Up to `MAX7219_MAX_CASCADE_SIZE` MAX721xx cascaded
    bool mirrored;               //!< true for horizontally mirrored displays
    bool bcd;
    bool flag;                   //Whether to buffer the SPI output
    uint8_t internal_buffer[64];
} max7219_t;
```