UncleRus/esp-idf-lib #237: Request a driver for 8channel i2c expander PCA9557

yourEOF, 2021-08-05:
**Device description**

The PCA9557 is a silicon CMOS circuit which provides parallel input/output expansion for 
SMBus and I2C-bus applications. The PCA9557 consists of an 8-bit input port register, 
8-bit output port register, and an I2C-bus/SMBus interface. It has low current consumption 
and a high-impedance open-drain output pin, IO0.

**Device information**

The product information and the data sheet can be found at:
https://datasheet.lcsc.com/lcsc/1810010521_NXP-Semicon-PCA9557PW,118_C141380.pdf

**Purchasing the product**

The URLs where you can buy the device are:
https://lcsc.com/product-detail/Interface-I-O-Expanders_NXP-Semicon-PCA9557PW-118_C141380.html

**Similar drivers**
As there is a driver for the 16channel expander PA95x5 would it be possible for you to make one for the 8bit PCA9557?

UncleRus, 2021-08-05:
Hi!

Please see this PR: #238

yourEOF, 2021-08-06:
Only tested input so far with your example. Works like a charm!
Thanks, never thought you would be so quick!
Best regards

commit 9ff12bcc (2021-08-06): feature: driver + example for PCA9557 (fixes #237)
commit a7c14b31 (2021-08-06): Merge pull request #238 from UncleRus/pca9557

feature: driver + example for PCA9557 (fixes #237)
closed 2021-08-06 by yourEOF