UncleRus/esp-idf-lib #241: PCA9537 : 4-bit I2C-bus and SMBus low power I/O por

SERIDJ, 2021-08-09:
I am requesting a driver for `PCA9537 `.

### Device description

`PCA9537DP ` is a chip to do a great thing. It supports only I2C. It
supports VDD from 2.3V to 5.5V. You can use it for saving the world.

### Device information

The product information and the data sheet can be found at:

https://github.com/ControlEverythingCommunity/PCA9537-02

The URLs where you can buy the device are:

- [ ] Farnell : https://fr.farnell.com/nxp/pca9537dp-118/ic-io-port-i2c-smbus-4bit-10tssop/dp/2212094RL?cjevent=467cf9a5f90911eb833302330a180510&cjdata=MXxZfDB8WXww&CMP=AFC-CJ-FR-1765328&gross_price=true&source=CJ
- [ ] Digi-Key : https://www.digikey.fr/product-detail/fr/nxp-usa-inc/PCA9537DP-118/568-1837-2-ND/789926

UncleRus, 2021-08-10:
I added support for this chip to the driver for pca9557. The changes are already in master, please test.

closed 2021-08-10 by UncleRus

SERIDJ, 2021-08-11:
@UncleRus  thanks, i will do that when i have time and let u know for result

SERIDJ, 2021-08-12:
@UncleRus  hi i cant find this reference PCA9557 to shop device or kit dev ant footprint is TSSOP10 so if u can we change it with    PCA9533DP/02,118 its better for me to buy and test,

let me know if i should create another requesting ...  ?

commit a1f4c046 (2021-08-10): pca9557: add support for PCA9537 (fixes #241)