UncleRus/esp-idf-lib #251: CI: "Invalid certificate" error

UncleRus, 2021-10-01:
CI build for esp-idf v4.x fails with error
> gen_crt_bundle.py: Invalid certificate in /home/runner/work/esp-idf-lib/esp-idf-lib/idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem

See https://github.com/UncleRus/esp-idf-lib/runs/3765359930

@trombik Could you take a look at this?

trombik, 2021-10-01:
ack

limpens, 2021-10-01:
Have a look at this: https://github.com/espressif/esp-idf/issues/7621#issuecomment-932070666

It looks like the issue is concentrating around the expired "DST Root CA X3", but the culprit looks to be the EC-ACC certificate.

Hope this helps.

trombik, 2021-10-02:
probably, disabling CONFIG_MBEDTLS_CERTIFICATE_BUNDLE should work, but I'm wondering how the upstream resolve the issue.

trombik, 2021-10-06:
updates from the upstream:

https://github.com/espressif/esp-idf/issues/7631#issuecomment-934212224

PR #283 (2022-01-22): ci: remove workaround for wrong CA bundle
fixed in 4.2, 4.3, and master

https://github.com/espressif/esp-idf/issues/7631#issuecomment-948445140

fixes #251
commit 5b8c8192 (2022-01-22): ci: remove workaround for wrong CA bundle

fixed in 4.2, 4.3, and master

https://github.com/espressif/esp-idf/issues/7631#issuecomment-948445140

fixes #251
closed 2022-01-22 by UncleRus