UncleRus/esp-idf-lib #255: cannot read ds18b20 temp

zenz, 2021-11-01:
@UncleRus 
Hi, It seems the ds18x20 lib cannot read temperature while using esp8266_rtos_sdk v3.4
no matter ds18x20_multi or ds18x20_single gets empty value like below.
```
I (1019) ds18x20_test: Sensor ffffffffffffffff: °C
I (2769) ds18x20_test: Sensor ffffffffffffffff: °C
```

By the way. while using esp8266_rtos_sdk 3.4, the GPIO pin must be init before measure. Or else it cannot find sensors.

_Originally posted by @zenz in https://github.com/UncleRus/esp-idf-lib/issues/58#issuecomment-956077125_

zenz, 2021-11-01:
OK, it's because the newlib defaults to nano mode, make menuconfig to swithc it to normal mode, then it comes back.

closed 2021-11-01 by zenz

yuanminglove, 2025-05-20:
yeah