UncleRus/esp-idf-lib #260: a component should have component meta data

trombik, 2021-11-11:
**Device type**:

- [x] ESP32
- [x] ESP8226

**Framework version**:

does not really matter.

**Describe the bug**:

Each component should have meta data. the meta data can be used for automation, and,  possibly, other things.

an example meta data file for `ads111x` (`components/ads111x/.eil.yml`) looks like this:

```yaml
---
components:
  - name: foo
    group: ADC/DAC
    licenses:
      - name: ISC
    targets:
      - name: esp32
    description: foo driver
  - name: ads111x
    description: |
      Driver for ADS1113/ADS1114/ADS1115 and ADS1013/ADS1014/ADS1015 I2C ADC
    group: ADC/DAC
    groups: []
    code_owners:
      - name: UncleRus
    depneds:
      - name: i2cdev
      - name: log
      - name: esp_idf_lib_helpers
    thread_safe: yes
    targets:
      - name: esp32
      - name: esp8266
    licenses:
      - name: BSD-3-Clause
    copyrights:
      - name: UncleRus
        full_name: Ruslan V. Uss
        email: unclerus@gmail.com
        year: 2016
      - name: dianlight
        full_name: Lucio Tarantino
        year: 2020
```
for starter, it helps to generate `README.md`. `Components` section would look like this:

```erb
## Components
<%
    groups = []
    @components.each do |c| 
      c["components"].each do |sub_c|
        groups << sub_c["group"]
      end 
    end 
    groups = groups.uniq.sort
-%>
<% groups.each do |g| %>
### <%= g %> libraries

| Component      | Description                                                             | License | ESP8266 | Thread safety
|----------------|-------------------------------------------------------------------------|---------|---------|---------------
<%
  group_components = []
  @components.each do |component|
    group_components += component["components"].select { |sub_component| sub_component["group"] == g } 
  end 
  group_components.each do |c| 
    name = c["name"]
    license = c["licenses"].map { |l| l["name"] }.join(", ")
    targets = c["targets"].map { |t| t["name"] }
    esp8266_support = targets.include?("esp8266") ? "Yes" : "No"
    thread_safe = c["thread_safe"] == "yes" ? "Yes" : "No"
    description = c["description"].chomp -%> 
| **<%= name %>** | <%= description %> | <%= license %> | <%= esp8266_support %> | <%= thread_safe %>
<% end -%> 
<% end -%> 
```

the meta data file format is not yet fixed.

other possibilities include:

* skip compilation of some components if they are not needed
* ensure the license is valid
* ensure the copyright holders are properly attributed
* clarify code owner
* generate other meta data

trombik, 2021-11-11:
@UncleRus if you like the idea, I'll implement it

UncleRus, 2021-11-11:
I love the idea.
Next step is generating documentation source, not just a README.

nicola-lunghi, 2021-11-16:
@UncleRus @trombik 
Have a look here:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html

I think espressif is developing their own "component library" maanager.
It would be useful to implement their metadata instead

trombik, 2021-11-18:
fixed in #261

closed 2021-11-18 by trombik

pedrominatel, 2021-11-24:
> @UncleRus @trombik Have a look here:
> 
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html
> 
> I think espressif is developing their own "component library" manager. It would be useful to implement their metadata instead

Hi, we (Espressif) are interested in having the libraries on the Components Manager. This will be an extremely valuable contribution to the community!

Can we talk about it @UncleRus?

Components Manager: https://components.espressif.com/search/

We can help you on this process.

trombik, 2021-11-24:
as the author of the issue and the pr, I would like to emphasize the purposes. the issue was raised to fix problems we have. updating README is tedious and error prone. consistency is not guaranteed. code _might_ have GPL code. I'd like to fix these issues first. on the other hand, few users requested to support third-party repositories, and the possible benefits they claimed have not convinced me. most of users are not beginners. they know how to install the libraries, to use git, or to try a proposed fix branch in their project. however, I'm not against the idea only if it's not another burden for us.

once we have stable metadata, it would be a matter of simple conversion to support your project. however, there are other issues. version management and release management. that's where we could break user projects, which I would like to avoid. I don't think you can help us here. 

I've already had a look at your implementation and was happy to see you are implementing `bundler`-like behaviour. we should not repeat same failures of `npm`.

trombik, 2021-11-24:
here is one thing you can help: clearly stating espressif policy about trademark infringement. I asked i_grr about this on twitter. public statement will certainly help. if you're curious why it matters, ask Google about leftpad and Kik.

trombik, 2022-02-09:
at the moment, the component repository is alpha. see https://github.com/espressif/idf-component-manager/issues/4 for details. i will see how it works and benefits us when it's ready.