UncleRus/esp-idf-lib #265: I2C Frequency Setting for BME680 Driver

tllaughn, 2021-11-22:
In the components/bme680/bme680.c file, the frequency is defined as 
#define I2C_FREQ_HZ 1000000 // Up to 3.4MHz, but esp-idf only supports 1MHz

However, for the ESP32C3 (and perhaps other platforms), the frequency should only be 100kHz.
#define I2C_FREQ_HZ 100000 // Up to 3.4MHz, but esp-idf only supports 1MHz

UncleRus, 2021-11-28:
Yes, I suppose it would be nice to limit the default frequency for different platforms.

But for now you easily can set whichever I2C clock you want [FAQ](https://github.com/UncleRus/esp-idf-lib/blob/master/FAQ.md#how-can-i-change-frequency-of-i2c-clock-at-default-frequency-my-device-is-unstable-or-not-working-at-all).

tllaughn, 2021-11-28:
Awesome!  Thanks.

Please forgive any typos or autocorrections!
-Theo

trombik, 2022-03-30:
@tllaughn are you sure the max frequency for esp32c3 is 100 Khz? the doc says it supports 1 Mhz at least, and 2 Mhz with `XTAL` as clock source.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/i2c.html

trombik, 2022-03-30:
i also thought per-device-per-target `I2C_FREQ_HZ` would be nice to have, but max frequency depends on device, target, _and_ clock source. that means it depends on how users use the target. as the FAQ suggests, you should change the frequency in your code.

changing the default to the fastest frequency for all targets is another idea, but I'm not sure.

closed 2022-05-29 by UncleRus