UncleRus/esp-idf-lib #270: Unnecessary delay in led_strip_flush?

dizcza, 2021-12-08:
Hi,
I'm looking into the led_strip module and considering using it in an application for M5Core2. The application requires the precision of tens of microseconds.
1. I've spotted the delay `ets_delay_us(50)` in the `led_strip_flush` function introduced in https://github.com/UncleRus/esp-idf-lib/commit/4642ea38826b2f70e1cfcb979544aa95a266f5a7. It looks to be an unnecessary delay since the previous function, `rmt_wait_tx_done`, takes care of the delay. I mean, if `rmt_wait_tx_done` results in `ESP_OK`, there is no need to wait further, isn't it? And the function returns an error otherwise. Was `ets_delay_us(50)` left after debugging or there is a reason behind it?
2. Since I set `CONFIG_LED_STRIP_FLUSH_TIMEOUT` to zero, would it make sense to use the LED strip at all? I mean, would it work... in theory?

UncleRus, 2021-12-08:
Hi!

1. This 50 us delay is _between_ the sending full data to the all LEDs in strip. Without this delay, a bug with "missing" pixels is possible: if delay between calls to `led_strip_flash()` is small, the LEDs consider the new data package sent to all LEDs in strip to be a continuation of the previous one.
2. This parameter is passed directly to the `rmt_wait_tx_done()` function. If you made it zero and the RMT channel is busy, instead of waiting for the end of the data transfer on the RMT channel and starting a new transmission, `led_strip_flush()`will simply return `ESP_ERR_TIMEOUT`.

Please ask questions in the "Discussions" section in the future.

closed 2021-12-08 by UncleRus

dizcza, 2021-12-09:
Thanks for the answer, and sorry, I've not noticed the Discussion section, I'll ask questions there in the future.

But what if the user calls `led_strip_flash` not so often, say each second, such that no delay between calls is needed? Without the delay, `led_strip_flash` takes ~45 microseconds. Adding 50 us doubles the runtime for nothing!

Could we perhaps add the `CONFIG_LED_STRIP_FLUSH_RMT_WAIT` option to specify the number of microseconds to wait after the completion of the RMT transmission? And comment in the description just what you've said.

UncleRus, 2021-12-09:
Done.

commit 4fb0e6e2 (2021-12-09): led_strip: fixes #270