UncleRus/esp-idf-lib #273: I always get reset when using i2c libraries like sht3x, sht4x, aht...

hoatenhu, 2021-12-25:
**Device type**:

- [x] ESP32

**Framework version**:

- [x] ESP-IDF 4.1

**Describe the bug**:

Here is my log, 

my chip is reset all the time (esp32-wroom-32)

[image]

hoatenhu, 2021-12-25:
when i remove ESP-ERR-CHECK in my code it gives this error

[image]

UncleRus, 2021-12-25:
Please read the FAQ:
https://github.com/UncleRus/esp-idf-lib/blob/master/FAQ.md#how-to-debug-i2c-based-drivers
This is I2C scanner:
https://github.com/UncleRus/esp-idf-i2cscan

PS I kindly ask you not to use images when you post logs. Use markdown instead.

closed 2021-12-25 by UncleRus