UncleRus/esp-idf-lib #276: ESP_IDF_VERSION is now in ESP8266 RTOS SDK version 3.4

trombik, 2022-01-05:
**Device type**:

- [] ESP32
- [x] ESP8226

**Framework version**:

- [] ESP-IDF latest master
- [] ESP-IDF 3.2.x
- [] ESP-IDF 3.3.x
- [x] ESP8266 RTOS SDK latest master
- [x] ESP8266 RTOS SDK 3.4

**Describe the bug**:

`ESP_IDF_VERSION`, and other related variables, such as `ESP_IDF_VERSION_MAJOR`, and `ESP_IDF_VERSION_VAL`, are included in ESP8266 RTOS SDK version 3.4. it is now possible to remove `esp_idf_lib_helpers` if all `esp-idf` versions have them.

as always, there is a catch: `make/version.mk` and `cmake/version.cmake` state that it is `4.0.0`.

UncleRus, 2022-01-09:
On the one hand, I've just revised the esp_idf_lib_helper code and it seems pretty appropriate to me:

- Firstly, this header defines handy macros like `HELPER_TARGET_ESP32`.
- Secondly, it will be very convenient to add to it any global workarounds for the entire ESP-IDF, which, I am sure, will be needed in the future.

On the other hand, getting rid of unnecessary dependency is always great.

So far, the arguments for leaving esp_idf_lib_helper as is seem more weighty to me.

trombik, 2022-01-23:
agreed.

closed 2022-01-23 by trombik