UncleRus/esp-idf-lib #298: scd4x: scd4x_get_data_ready_status not working

rale, 2022-02-27:
### The issue

In get_data_ready_status, data_ready is always being set to true, even when not actually ready.

From the datasheet:
`If the least significant 11 bits of word[0] are 0 → data not ready else → data ready for read-out`

I believe it should be something like:
`*data_ready = (status & 0x07ff);`

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

v4.4

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [X] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

scd4x

### Confirmation

- [X] This report is not a question nor a request for drivers.

UncleRus, 2022-02-27:
Thank you! Could you test the fix?

rale, 2022-02-27:
Yep, working correctly now.  Thanks