UncleRus/esp-idf-lib #300: CI randomly failing builds

UncleRus, 2022-03-03:
### The issue

Something wrong with CI: while building examples for esp32 it's randomly showing error:
```
In file included from /home/runner/work/esp-idf-lib/esp-idf-lib/components/mhz19b/mhz19b.c:48:
/home/runner/work/esp-idf-lib/esp-idf-lib/components/mhz19b/mhz19b.h:47:10: fatal error: driver/uart.h: No such file or directory
 #include <driver/uart.h>
          ^~~~~~~~~~~~~~~
compilation terminated.

```

### Confirmation

- [X] This report is not a bug report in code, a question, nor a request for drivers.

trombik, 2022-03-03:
https://github.com/UncleRus/esp-idf-lib/pull/299/commits/58bac5ca9b5d84b52931fe5c4421143553c38a88 should fix it.

UncleRus, 2022-03-03:
Oops, overlooked the obvious. Thanks!

UncleRus, 2022-03-03:
cherry-picked [58bac5c](https://github.com/UncleRus/esp-idf-lib/pull/299/commits/58bac5ca9b5d84b52931fe5c4421143553c38a88)

closed 2022-03-03 by UncleRus