UncleRus/esp-idf-lib #322: Does not buiild with ESP8266-RTOS-IDF

HankB, 2022-04-21:
### The issue

Compile errors building the library.

```text
CC build/i2cdev/i2cdev.o
In file included from /home/hbarta/esp/esp-idf-lib/components/i2cdev/i2cdev.h:40,
                 from /home/hbarta/esp/esp-idf-lib/components/i2cdev/i2cdev.c:37:
/home/hbarta/esp/esp-idf-lib/components/i2cdev/i2cdev.c: In function 'i2cdev_done':
/home/hbarta/esp/esp-idf-lib/components/i2cdev/i2cdev.c:53:62: error: 'CONFIG_I2CDEV_TIMEOUT' undeclared (first use in this function); did you mean 'CONFIG_FATFS_TIMEOUT_MS'?
         if (!xSemaphoreTake(states[port].lock, pdMS_TO_TICKS(CONFIG_I2CDEV_TIMEOUT))) \
                                                              ^~~~~~~~~~~~~~~~~~~~~

.
.
.
```

More at <https://pastebin.com/BF2WVbmu>

### Which SDK are you using?

ESP8266 RTOS SDK

### Which version of SDK are you using?

esp-2020r3-49-gd5524c1 (IDF version: v3.4-59-gbbde375b-dirty)

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [X] esp8266
- [ ] other

### Component causing the issue

esp-idf-lib/components/i2cdev/i2cdev.c

### Anything in the logs that might be useful for us?

```console
no logs, see compiler output at https://pastebin.com/BF2WVbmu
```

### Additional information or context

Thank you for your efforts to provide this library! My expectation is that it would work with the ESP8266-RTOS-IDF. Please advise if this is not the case. (I found perhaps a related issue https://github.com/UncleRus/esp-idf-lib/issues/46 that may have been the result of using PlatformIO but I am not using that in this case.)

Library is included in my project <https://github.com/HankB/ESP8266_RTOS_BME280>

Do not hesitate to ask if further information is needed or there is something for me to try.

### Confirmation

- [X] This report is not a question nor a request for drivers.

trombik, 2022-04-22:
what would happen if you compile an example?

```console
cd examples/bmp280
make menuconfig
make
```

HankB, 2022-04-23:
After setting ESP_IDF  (pointing to ESP8266_RTOS_IDF) and executing `. $ESP_IDF/export.sh` I switched too the repo directory and ran the commands you suggested. The build ran for a while until it reported

```text
make: *** No rule to make target 'component-driver-build', needed by 'component-led_strip-build'.  Stop.
```

trombik, 2022-04-23:
what if you update the ESP8266 RTOS SDK to the latest master?

HankB, 2022-04-23:
With an update to the ESP8266 RTOS SDK my project now builds the library without reporting any errors. For reference, the update reported:

```text
hbarta@olive:~/esp/ESP8266_RTOS_SDK$ git pull --recurse
Fetching submodule components/coap/libcoap
Fetching submodule components/coap/libcoap/ext/tinydtls
Fetching submodule components/json/cJSON
Fetching submodule components/lwip/lwip
From https://github.com/espressif/esp-lwip
   76303df2..fb1f3552  2.1.2-esp  -> origin/2.1.2-esp
Fetching submodule components/mbedtls/mbedtls
Fetching submodule components/mqtt/esp-mqtt
From https://github.com/espressif/esp-mqtt
   5878e2e..59b48a4  master     -> origin/master
Already up to date.
hbarta@olive:~/esp/ESP8266_RTOS_SDK$
```

Thank you for your help.

closed 2022-04-23 by HankB

trombik, 2022-04-23:
unfortunately, ESP8266 RTOS SDK is not like `esp-idf`. they rarely release, does not maintain issues or PRs. we do not support a specific release, but `master`, because of it. please always use `master`. the components are built in our CI with `master`.