UncleRus/esp-idf-lib #331: led_strip: Deprecated RMT driver

UncleRus, 2022-05-29:
### The issue

In the last ESP-IDF master, if the CONFIG_RMT_SUPPRESS_DEPRECATE_WARN configuration option is not set, a warning is displayed
```
#warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h".
```
Consideration should be given to migrating from the deprecated driver to the newer one in led_strip, while retaining the use of the older driver for compatibility with supported ESP-IDF versions.

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

master

### Which build target have you used?

- [X] esp32
- [X] esp32s2
- [X] esp32s3
- [X] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

led_strip

### Confirmation

- [X] This report is not a question nor a request for drivers.

PaulskPt, 2022-07-12:
I am trying to build the led_strip example for a M5Stam-C3. I get the error file "rmt_tx.h" not found. I did a search (using 'find') in my Linux Ubuntu system however I couldn't find it. I googled for it. Same negative result.

swkang0513, 2023-05-10:
Is there any update?

silardg, 2023-11-04:
Still nothing on this?

Alfred1975, 2023-11-27:
The latest esp-idf shows the same warning `rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h``

lukedukeus, 2024-06-26:
just ran into this, commenting to get notified of any update

trombik, 2025-02-25:
Since v5.0, the driver is effectively broken.

> rmt_fill_tx_items, rmt_write_items and rmt_write_sample are removed. In the new driver, user needs to provide an encoder to "translate" the user data into RMT symbols.

See [Migration from 4.4 to 5.0 Peripherals](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/migration-guides/release-5.x/5.0/peripherals.html).

The driver will be disconnected from the CI. Until someone steps up to port it to newer RMT driver, it will be broken.

jkosharek, 2025-07-11:
Is there another option for WS2812 type devices on ESP32/Arduino?

closed 2025-08-01 by trombik