UncleRus/esp-idf-lib #402: CI: Optimization of the list of drivers incompatible with ESP8266

UncleRus, 2022-06-29:
### The issue

In the current CI build scripts, exists small problem related to the list of components that are incompatible with the ESP8266: this list (`EXCLUDED_COMPONENTS`), firstly, is a constant that can only be modified manually, and secondly, it is declared in two places.
The further, the more drivers for SPI devices will appear in the repository that will be incompatible with ESP8266 and frequent edits of this list in three places (including README.md) are very inconvenient.
I believe that we need, firstly, to put down full compatibility lists in the metadata of the components (now only esp32 and esp8266 are indicated almost everywhere, without mentioning esp32c3 and esp32s4), and secondly, based on this data, automatically generate lists of compatible components for use in CI and `README.md`

### Confirmation

- [X] This report is not a bug report in code, a question, nor a request for drivers.

trombik, 2022-06-30:
the current `Build examples` works like this:

1. setup the build environment
1. build all examples in a loop unless a special `ignore` file does not exist in the example directory
 
a possible solution:

1. setup the build environment
2. build all examples in a loop if the metadata of a component matches the supported build targets

the problems:

1. as a component may have multiple examples, and examples do not have a metadata, it is necessary to build examples by component, not by example directory name
2. another convention for examples is necessary, such as examples for `components/foo` is `examples/foo/${EXAMPLE_NAME}`
3. parsing a metadata and writing conditions to build are not easy. there should be tests for the logic
4. JavaScript, or typescript, is not my language, and it would take some time to develop the solution.

trombik, 2022-06-30:
a possible issue:

the metadata contains a list of supported targets (whitelist). this is not ideal because the field must be updated manually as espressif releases a new product. this approach is `not-supported-unless-proven-to-work`.

an alternative is a blacklist, say, the component supports all targets unless proved otherwise. this is easier to maintain as it is not necessary to update the field whenever a new product is released. however, it is not possible to say the component is supported just by the results of the CI, as the CI cannot catch runtime errors.

trombik, 2022-07-03:
> as a component may have multiple examples, and examples do not have a metadata, it is necessary to build examples by component, not by example directory name

> another convention for examples is necessary, such as examples for components/foo is examples/foo/${EXAMPLE_NAME}

fixed in #411 

> parsing a metadata and writing conditions to build are not easy. there should be tests for the logic

this is not yet implemented.

PR #414 (2022-07-04): ci: remove IGNORE_FILE from the CI
fixes #402
commit 55c81496 (2022-07-04): ci: build EXCLUDE_COMPONENTS from metadata, skip EXCLUDE_COMPONENTS

in the build. remove IGNORE_FILE, which is no longer necessary.

fixes #402
closed 2022-07-04 by UncleRus