UncleRus/esp-idf-lib #438: Drive for Accelerometer and Gyroscope - MPU6050

hebersantos, 2022-07-26:
### Device name

MPU6050

### Device description

The MPU-6050 IMU (Inertial Measurement Unit) is a 3-axis accelerometer and 3-axis gyroscope sensor. The accelerometer measures the gravitational acceleration, and the gyroscope measures the rotational velocity. Additionally, this module also measures temperature. This sensor is ideal for determining the orientation of a moving object.

### Device product page URL

https://invensense.tdk.com/products/motion-tracking/6-axis/mpu-6050/

### Datasheet URL

https://invensense.tdk.com/wp-content/uploads/2015/02/MPU-6000-Datasheet1.pdf

### Can the device be purchased at any of the followings?

- [X] AliExpress
- [X] Amazon
- [X] ebay
- [X] Digi-Key
- [X] Mouser

### URLs to purchase the device

- https://pt.aliexpress.com/item/32346328217.html?spm=a2g0o.productlist.0.0.44bc64fdc5M8g5&algo_pvid=20660760-168b-448b-83f4-6f176cca6c50&algo_exp_id=20660760-168b-448b-83f4-6f176cca6c50-1&pdp_ext_f=%7B%22sku_id%22%3A%2254741193371%22%7D&pdp_npi=2%40dis%21BRL%21%218.95%21%21%2120.37%21%21%402101d4a716588432131917381e4723%2154741193371%21sea
- https://www.amazon.com/MPU-6050-MPU6050-Accelerometer-Gyroscope-Converter/dp/B07RXQGGJX/ref=sr_1_2?crid=PKBZ02H35KPU&keywords=mpu6050&qid=1658843548&sprefix=mpu6050%2Caps%2C183&sr=8-2
- https://www.ebay.com/itm/201234452818?hash=item2eda821152:g:qmkAAOSw7ThUgX1W&amdata=enc%3AAQAHAAAA0Fr3znTxwSX8OVMDylmCKYtHeK7a0r5eA2Zxm2rbGKJhGe9f9j57c4Sf404gppZEJX%2FHzOWU2S4bmsq0dycN%2BhOl8i49ehoS2f7yPK6cdpx0zWJX4LQGT4dCduGVPPZxN6WHgqS0fHdsVPiKUEQvcXvJbh7acpgG8zNEYNZUSIbD3jUU3ciNziScNHa7Qw2%2Bk1SNdU6Ll%2B%2BtWi5igoja8%2FzAd7CQ0dOrJEBTXVynuX9I7UecClL9APsXAiSOgITQBLmlBA3tMqTXD5cEyAOJXkM%3D%7Ctkp%3ABFBM2IfarMdg
- https://www.digikey.com.br/en/products/detail/dfrobot/SEN0142/6588492
- https://br.mouser.com/ProductDetail/DFRobot/SEN0142?qs=Zcin8yvlhnO0Rr0B1JJGiw%3D%3D

### Other implementations

https://randomnerdtutorials.com/esp32-mpu-6050-accelerometer-gyroscope-arduino/

horsemann07, 2022-08-20:
Hello @UncleRus @hebersantos 

Here, I try to convert the mpu6050 Arduino code for esp-idf, it is not as per your coding standard. I test with the device and it is working with esp-idf 4.4, [link](https://github.com/horsemann07/esp_idf_lib/tree/feature-implement-mpu6050)

I m trying to change the code as per the coding standard.

Thanks!

hebersantos, 2022-08-30:
Hello @horsemann07 @UncleRus 

I've beem busy with others projects this days and couldn't  focus on the project of the requested!
But Thanks a lot, I'll test and report here the results.
I'm using it integrated with others i2c sensors, so I'll test it too.

Thanks!

horsemann07, 2022-11-10:
> * Most importantly, your driver can only work with a single device. It is not possible to connect several sensors at the same time. All other drivers in this repository can work simultaneously with any number of devices of the same type connected to the esp. Please use the same approach.

Hello @UncleRus,

[Here](https://github.com/horsemann07/esp-idf-lib/tree/feature-implement-mpu6050/components/mpu6050), I tried to make a driver which works on several sensors of the same type with one esp32. But the system is crashing while reading the two i2c. It happening while taking the semaphore in i2c_dev. 

I m using different two different ports with different pin no. to read two mpu6050 sensors on 1 esp32. I saw some examples using `tca9548 ` components to work with multi i2c. 
But I think esp32 supports 3 ports so we can connect 3 sensors using other i2c multichannel components.
So, why the system is crashing taking semaphores of the different ports?

Can you help me with this? 

Here is the log and [example link](https://github.com/horsemann07/esp-idf-lib/blob/feature-implement-mpu6050/examples/mpu6050/multi_switch_i2c/main/main.c)
```

W (305) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (319) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (329) mpu6050_test: mpu6050 config 1: addr 0x68, sda 32, scl 33, clk, 100000 port 2
I (0) mpu60
assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))

Backtrace:0x4008192e:0x3ffb9a200x400863f1:0x3ffb9a40 0x4008b701:0x3ffb9a60 0x4008724e:0x3ffb9b80 0x400d576f:0x3ffb9bc0 0x400d5476:0x3ffb9be0 0x400d4f7e:0x3ffb9c10 0x40088f49:0x3ffb9c50 
0x4008192e: panic_abort at /home/raghav/Desktop/Personal/esp_idf_v4.4.1/components/esp_system/panic.c:402

0x400863f1: esp_system_abort at /home/raghav/Desktop/Personal/esp_idf_v4.4.1/components/esp_system/esp_system.c:128

0x4008b701: __assert_func at /home/raghav/Desktop/Personal/esp_idf_v4.4.1/components/newlib/assert.c:85

0x4008724e: xQueueSemaphoreTake at /home/raghav/Desktop/Personal/esp_idf_v4.4.1/components/freertos/queue.c:1545 (discriminator 1)

0x400d576f: i2c_dev_probe at /home/raghav/Desktop/Personal/esp_idf_v4.4.1/examples/personal/esp-idf-lib/components/i2cdev/i2cdev.c:244

0x400d5476: mpu6050_init_desc at /home/raghav/Desktop/Personal/esp_idf_v4.4.1/examples/personal/esp-idf-lib/components/mpu6050/mpu6050.c:2628 (discriminator 2)

0x400d4f7e: mpu6050_test1 at /home/raghav/Desktop/Personal/esp_idf_v4.4.1/examples/personal/esp-idf-lib/examples/mpu6050/multi_switch_i2c/build/../main/main.c:82

0x40088f49: vPortTaskWrapper at /home/raghav/Desktop/Personal/esp_idf_v4.4.1/components/freertos/port/xtensa/port.c:131

```

closed 2023-07-21 by UncleRus