UncleRus/esp-idf-lib #449: TSL2591 not working

rhr407, 2022-08-14:
### The issue

When running the example code, the following text keeps printing on the terminal.

```
entry 0x40080694
I (27) boot: ESP-IDF v4.4.2-74-g4fb74a808b-dirty 2nd stage bootloader
I (27) boot: compile time 10:24:20
I (28) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=09944h ( 39236) map
I (126) esp_image: segment 1: paddr=0001996c vaddr=3ffb0000 size=023a0h (  9120) load
I (129) esp_image: segment 2: paddr=0001bd14 vaddr=40080000 size=04304h ( 17156) load
I (138) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=17e5ch ( 97884) map
I (175) esp_image: segment 4: paddr=00037e84 vaddr=40084304 size=086f8h ( 34552) load
I (190) esp_image: segment 5: paddr=00040584 vaddr=50000000 size=00010h (    16) load
I (196) boot: Loaded app from partition at offset 0x10000
I (196) boot: Disabling RNG early entropy source...
I (210) cpu_start: Pro cpu up.
I (210) cpu_start: Starting app cpu, entry point is 0x40081110
0x40081110: call_start_cpu1 at /home/riz/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (224) cpu_start: Pro cpu start user code
I (224) cpu_start: cpu freq: 160000000
I (224) cpu_start: Application information:
I (229) cpu_start: Project name:     example-tsl2591_simple
I (235) cpu_start: App version:      0.9.1-6-g977da43-dirty
I (241) cpu_start: Compile time:     Aug 14 2022 10:28:20
I (247) cpu_start: ELF file SHA256:  f4d8c48edd3ef85a...
I (253) cpu_start: ESP-IDF:          v4.4.2-74-g4fb74a808b-dirty
I (260) heap_init: Initializing. RAM available for dynamic allocation:
I (267) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (273) heap_init: At 3FFB2CE8 len 0002D318 (180 KiB): DRAM
I (279) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (286) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (292) heap_init: At 4008C9FC len 00013604 (77 KiB): IRAM
I (300) spi_flash: detected chip: generic
I (303) spi_flash: flash io: dio
W (307) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (321) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Lux: nan
Lux: nan
Lux: nan
Lux: nan
Lux: nan
Lux: nan
Lux: nan
Lux: nan
Lux: nan
Lux: nan
Lux: nan
Lux: nan
Lux: nan
Lux: nan
Lux: nan

```

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

4.4.2

### Which build target have you used?

- [X] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

tsl2591

### Confirmation

- [X] This report is not a question nor a request for drivers.

UncleRus, 2022-08-14:
@juliandoerner Could you take a look at this?

rhr407, 2022-08-23:
Any update on this issue?

juliandoerner, 2022-08-23:
I will try to reproduce on at the weekend.

rhr407, 2022-08-27:
Any progress on this issue?

rhr407, 2022-08-29:
Working for me after adding 4 lines in the init function.

```
esp_err_t tsl2591_init(tsl2591_t *dev)
{
    CHECK_ARG(dev);
    ESP_LOGD(TAG, "Initialize sensor.");

    I2C_DEV_TAKE_MUTEX(&dev->i2c_dev);

    tsl2591_set_power_status(dev, TSL2591_POWER_ON);
    tsl2591_set_als_status(dev, TSL2591_ALS_ON);
    tsl2591_set_gain(dev, TSL2591_GAIN_MEDIUM);
    tsl2591_set_integration_time(dev, TSL2591_INTEGRATION_300MS);

    uint8_t tmp_reg = 0;
    I2C_DEV_CHECK(&dev->i2c_dev, read_enable_register(dev, &tmp_reg));
    dev->settings.enable_reg = tmp_reg;
    ESP_LOGD(TAG, "Initial enable register: %x.", tmp_reg);

    I2C_DEV_CHECK(&dev->i2c_dev, read_control_register(dev, &tmp_reg));
    dev->settings.control_reg = tmp_reg;
    ESP_LOGD(TAG, "Initial control register: %x.", tmp_reg);

    I2C_DEV_CHECK(&dev->i2c_dev, read_register(dev, TSL2591_REG_PERSIST, &tmp_reg));
    dev->settings.persistence_reg = tmp_reg;
    ESP_LOGD(TAG, "Initial persistence filter: %x.", tmp_reg);

    I2C_DEV_GIVE_MUTEX(&dev->i2c_dev);

    // Wait until the first integration cycle is completed.
    tsl2591_integration_time_t integration_time;
    ESP_ERROR_CHECK(tsl2591_get_integration_time(dev, &integration_time));
    switch (integration_time)
    {
        case TSL2591_INTEGRATION_100MS:
            SLEEP_MS(110);
            break;
        case TSL2591_INTEGRATION_200MS:
            SLEEP_MS(210);
            break;
        case TSL2591_INTEGRATION_300MS:
            SLEEP_MS(310);
            break;
        case TSL2591_INTEGRATION_400MS:
            SLEEP_MS(410);
            break;
        case TSL2591_INTEGRATION_500MS:
            SLEEP_MS(510);
            break;
        case TSL2591_INTEGRATION_600MS:
            SLEEP_MS(610);
            break;
    }

    return ESP_OK;
}
```

closed 2022-08-29 by rhr407