UncleRus/esp-idf-lib #454: Driver for am2320 using i2c

aschiavon91, 2022-09-23:
### Device name

am2320

### Device description

AM2320 is a temperature and humidity composite sensor with a calibrated digital signal output

### Device product page URL

http://www.aosong.com/en/products-41.html

### Datasheet URL

https://cdn-shop.adafruit.com/product-files/3721/AM2320.pdf

### Can the device be purchased at any of the followings?

- [X] AliExpress
- [ ] Amazon
- [ ] ebay
- [ ] Digi-Key
- [ ] Mouser

### URLs to purchase the device

https://aliexpress.com/item/32654739552.html?spm=a2g0o.productlist.0.0.59477cb0e9ifXo&algo_pvid=ecacd069-bea0-4953-8238-ccf3fb3de654&algo_exp_id=ecacd069-bea0-4953-8238-ccf3fb3de654-0
https://aliexpress.com/item/1005002884884196.html

### Other implementations

https://github.com/adafruit/Adafruit_AM2320
https://github.com/Ten04031977/AM2320-master/blob/master/am2320.cpp

JaimeAlbq, 2022-10-19:
Hi all,

I do not have the hardware but I could do the port of this Adafruit's library.

@aschiavon91, would you be able to test it?

aschiavon91, 2022-10-20:
for sure @JaimeAlbq!

sirapol, 2022-11-03:
I still waiting on this issue.
I try DHT lib (AM2321), It not compatible.
Because, AM2321 has 2 protocol ( I2C and one wire ).

trombik, 2022-11-03:
I ordered a few last month. the tracking information shows it's been stacked at custom.

sirapol, 2022-11-03:
I can using AM2320 with EPS-IDF-LIB ( using DHT lib ).
AM2320 has 2 protocol.
1. I2C ( Pin 1:VCC , 2:SDA , 3:GND , 4:SCL) this mode not compatible with DHT lib.
2. Single bus ( Pin 1:VCC, 2:Data ,3:GND , 4:GND) this mode working with DHT lib.

you can using this example : https://github.com/UncleRus/esp-idf-lib/blob/master/examples/dht/default/main/main.c
And just define to DHT_TYPE_AM2301.
And just define data pin on line 26.

I hope will help you.

By the way, I tried to using I2C mode many time.
I think I2C on ESP-IDF very complex to using.
Please tell me step by step for using.
Thank you.

trombik, 2022-11-03:
@sirapol the i2c component in esp-idf is far better than one in Arduino. well documented, reliable, and clean. if you don't like it, that's your choice. you don't have to use it. read the official documentation, read the source code of the i2cdev component. full of examples.

aschiavon91, 2022-11-12:
@sirapol I really appreciate the help, I'm sure this could help someone else searching for this.
Just for info, I already using AM2320 through one wire proto, but in addition to what @trombik said, i2c could also help to avoid complex wiring, for projects with multiple sensors.

UncleRus, 2023-04-20:
Please test #508

PR #508 (2023-06-02): Driver+example for AM2320 (I2C mode)
Fixes #454

@aschiavon91, @sirapol, @pfuentes69 Please test
closed 2023-06-02 by UncleRus