UncleRus/esp-idf-lib #457: Shift Register

JaimeAlbq, 2022-10-09:
### Device name

Shift Register

### Device description

Shift register to expend I/O by writing or reading from it in a serial communication.
In this case I am considering the IC 74HC595.

**NOTE**: The driver can and will enhance other drivers, such as [hd44780](/UncleRus/esp-idf-lib/tree/master/components/hd44780).

### Device product page URL

https://www.ti.com/product/SN74HC595

### Datasheet URL

https://www.ti.com/lit/ds/symlink/sn74hc595.pdf

### Can the device be purchased at any of the followings?

- [X] AliExpress
- [X] Amazon
- [X] ebay
- [X] Digi-Key
- [X] Mouser

### URLs to purchase the device

[AliExpress](https://pt.aliexpress.com/item/32770496442.html?spm=a2g0o.productlist.0.0.178031d9bUpkjk&algo_pvid=60150d89-4e84-4a53-8b2f-aa2d95e508e9&algo_exp_id=60150d89-4e84-4a53-8b2f-aa2d95e508e9-1&pdp_ext_f=%7B%22sku_id%22%3A%2262383502123%22%7D&pdp_npi=2%40dis%21BRL%214.44%214.44%21%21%2128.85%21%21%40210318d116652735155986427e8a7d%2162383502123%21sea&curPageLogUid=yyCwE5y3ppoV)
[Amazon](https://www.amazon.com/MCIGICM-74HC595-SN74HC595N-Register-registers/dp/B07HFWB9L9/ref=sr_1_2?keywords=74hc595&qid=1665273467&qu=eyJxc2MiOiIzLjc1IiwicXNhIjoiMy4wNCIsInFzcCI6IjIuNjkifQ%3D%3D&sr=8-2)
[ebay](https://www.ebay.com/itm/203225096052?hash=item2f5128df74:g:6HYAAOSwE-df4AoZ&amdata=enc%3AAQAHAAAA4IJ%2BLTYOWfYmUybWj%2FJESL%2FJ%2FeHBeU4gJ7WnFxYr4ADfm9oMJAkRvPA2X6PELxHUTnk78kyCnyDLlNKMOe%2B99kiK40L7EwkR2gSCEU0a%2Fi%2FbjnjCqPIUcezSsMRdcUS%2Fsw22W1IWpgzl2vD14s1hkLQm4cLntbAJ3o4uPjy7fVlOQnN5IwTCShePc9eSXWcHtQGVdyEPBQfWfq0s6Oyasdpr2c8JMAJHESab5HnStwXCsXshDdAZDFkzkD2V3NogzedssZKLvxQcyiJHKuBnm3S9UoO4P%2Be7akidzwPo5KHk%7Ctkp%3ABk9SR7qrkqD3YA)
[Digi-Key](https://www.digikey.com/en/products/detail/texas-instruments/SN74HC595DT/1571266)
[Mouser](https://www.mouser.com/ProductDetail/Texas-Instruments/CD74HC595M?qs=xFfolx0DHx1cUZwG%2F8WRcQ%3D%3D)

### Other implementations

[Arduino - shiftOut()](https://www.arduino.cc/reference/en/language/functions/advanced-io/shiftout/)

trombik, 2022-10-09:
do you have a plan to work on a generic shift resister driver based on this?
 
https://github.com/JaimeAlbq/ic74hc595

JaimeAlbq, 2022-10-09:
> do you have a plan to work on a generic shift resister driver based on this?
> 
> https://github.com/JaimeAlbq/ic74hc595

Hello @trombik,

Actually I already did the implement with some improvements. But basically will be very similar as the mentioned repository.

Do you have any suggestions?

trombik, 2022-10-09:
create a PR. we will help you by reviewing. some generic advice:

* read other driver code, learn the style
* read CONTRIBUTION.md
* don't forget esp8266 and `make` build method

JaimeAlbq, 2022-10-11:
Hi @trombik,

I have finished the implemention of the shift register feature, now I am working on the integration between it and the [hd44780](https://github.com/UncleRus/esp-idf-lib/tree/master/components/hd44780). Do I need to create a new issue to perform the Pull Request or is there another way?

trombik, 2022-10-11:
@JaimeAlbq better to work on it after reviews.

trombik, 2025-08-01:
moved to: https://github.com/esp-idf-lib/core/issues/35