UncleRus/esp-idf-lib #464: SCD30 Example Doesn't Function with SCD30 & esp32s3

keenanjohnson, 2022-10-25:
### The issue

Hello! I'm trying to flash the scd30 example program to an esp32s3 board.

https://github.com/UncleRus/esp-idf-lib/tree/master/examples/scd30/default

I'm able to compile the example, and flash successfully. 

However, when I open the monitor, I see no output using the idf.py monitor command.

Any ideas for how to troubleshoot this futher? Has anyone else successfully run this example?

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

0.9.1

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [X] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

scd30

### Confirmation

- [X] This report is not a question nor a request for drivers.

UncleRus, 2022-10-25:
What OS are you using?
It looks like `idf.py monitor` is opening the wrong serial port.

keenanjohnson, 2022-10-25:
When I open either idf.py monitor or via screen, I don't receive any output.

UncleRus, 2022-10-26:
It's very strange. You should get at least bootloader output on `idf.py monitor`... Can you try another another chip?

trombik, 2022-10-26:
@keenanjohnson remove `sdkconfig`. run `idf.py menuconfig` and `idf.py flash monitor`. try `cu -s 115200 -l /dev/cu.usbmodem1101` as well. make sure that you have not accidentally modified the example. see `git status` output.

trombik, 2022-10-26:
if your development board has two USB ports, make sure you are using the correct port for serial communication.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html

keenanjohnson, 2022-10-26:
Thank you for the help as I am still learning esp-idf. I will try this and report back!

keenanjohnson, 2022-10-27:
FYI I do see the bootloader when I run the monitor. Here is my output of the monitor program: 

```
keenan@Keenans-BlueBook-Pro default % /Users/keenan/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/keenan/esp/esp-i
df/tools/idf.py -p /dev/cu.usbmodem1101 monitor
Executing action: monitor
Running idf_monitor in directory /Users/keenan/Documents/GitHub/esp-idf-lib/examples/scd30/default
Executing "/Users/keenan/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/keenan/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbmodem1101 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /Users/keenan/Documents/GitHub/esp-idf-lib/examples/scd30/default/build/example-scd30.elf -m '/Users/keenan/.espressif/python_env/idf4.4_py3.10_env/bin/python' '/Users/keenan/esp/esp-idf/tools/idf.py' '-p' '/dev/cu.usbmodem1101'"...
--- idf_monitor on /dev/cu.usbmodem1101 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a76
waiting for download
```

keenanjohnson, 2022-10-27:
I'm using this esp32s3 board from adafruit. https://www.adafruit.com/product/5477

Looks like the I2C pins may be different. I've tried switching them in the sdkconfig, but no different results in behavior. 

[image]

trombik, 2022-10-27:
> waiting for download

which means the bootloader is in download mode, not running the application. make sure `GPIO0` is HIGH.

UncleRus, 2022-10-29:
See also: #467