UncleRus/esp-idf-lib #477: onewire on espIDFv5 ets_sys.h: No such file. use esp_rom_delay_us exept ets_delay_us

politsin, 2022-11-15:
### The issue

## I got en error
```
ets_sys.h: No such file
```

In the 5v esp-idf  https://github.com/espressif/esp-idf/releases/tag/v5.0-beta1
in `Other` sectinon we can see:
ROM: rom//ets_sys.h are not exposed from public header files by default.

## Solution
1) Replace all `ets_delay_us` whith `esp_rom_delay_us ` in `onewire.c`
2) add `esp_rom` as dependency

```diff
diff --git a/components/onewire/CMakeLists.txt b/components/onewire/CMakeLists.txt
index d89d5f5..a00c39e 100644
--- a/components/onewire/CMakeLists.txt
+++ b/components/onewire/CMakeLists.txt
@@ -1,7 +1,7 @@
 if(${IDF_TARGET} STREQUAL esp8266)
     set(req esp8266 freertos esp_idf_lib_helpers)
 else()
-    set(req driver freertos esp_idf_lib_helpers)
+    set(req driver freertos esp_idf_lib_helpers esp_rom)
 endif()
 
 idf_component_register(
diff --git a/components/onewire/component.mk b/components/onewire/component.mk
index 877b269..3f3cf02 100644
--- a/components/onewire/component.mk
+++ b/components/onewire/component.mk
@@ -3,5 +3,5 @@ COMPONENT_ADD_INCLUDEDIRS = .
 ifdef CONFIG_IDF_TARGET_ESP8266
 COMPONENT_DEPENDS = esp8266 freertos esp_idf_lib_helpers
 else
-COMPONENT_DEPENDS = driver freertos esp_idf_lib_helpers
+COMPONENT_DEPENDS = driver freertos esp_idf_lib_helpers esp_rom
 endif
diff --git a/components/onewire/onewire.c b/components/onewire/onewire.c
index 88d2ccb..0173e15 100644
--- a/components/onewire/onewire.c
+++ b/components/onewire/onewire.c
@@ -52,7 +52,7 @@
 #include <string.h>
 #include <freertos/FreeRTOS.h>
 #include <freertos/task.h>
-#include <ets_sys.h>
+// #include <ets_sys.h>
 #include <esp_idf_lib_helpers.h>
 #include "onewire.h"
 
@@ -84,12 +84,12 @@ static inline bool _onewire_wait_for_bus(gpio_num_t pin, int max_wait)
     {
         if (gpio_get_level(pin))
             break;
-        ets_delay_us(5);
+        esp_rom_delay_us(5);
     }
     state = gpio_get_level(pin);
     // Wait an extra 1us to make sure the devices have an adequate recovery
     // time before we drive things low again.
-    ets_delay_us(1);
+    esp_rom_delay_us(1);
     return state;
 }
 
@@ -115,11 +115,11 @@ bool onewire_reset(gpio_num_t pin)
         return false;
 
     gpio_set_level(pin, 0);
-    ets_delay_us(480);
+    esp_rom_delay_us(480);
 
     PORT_ENTER_CRITICAL;
     gpio_set_level(pin, 1); // allow it to float
-    ets_delay_us(70);
+    esp_rom_delay_us(70);
     bool r = !gpio_get_level(pin);
     PORT_EXIT_CRITICAL;
 
@@ -138,17 +138,17 @@ static bool _onewire_write_bit(gpio_num_t pin, bool v)
     if (v)
     {
         gpio_set_level(pin, 0);  // drive output low
-        ets_delay_us(10);
+        esp_rom_delay_us (10);
         gpio_set_level(pin, 1);  // allow output high
-        ets_delay_us(55);
+        esp_rom_delay_us (55);
     }
     else
     {
         gpio_set_level(pin, 0);  // drive output low
-        ets_delay_us(65);
+        esp_rom_delay_us (65);
         gpio_set_level(pin, 1); // allow output high
     }
-    ets_delay_us(1);
+    esp_rom_delay_us (1);
     PORT_EXIT_CRITICAL;
 
     return true;
@@ -161,11 +161,11 @@ static int _onewire_read_bit(gpio_num_t pin)
 
     PORT_ENTER_CRITICAL;
     gpio_set_level(pin, 0);
-    ets_delay_us(2);
+    esp_rom_delay_us (2);
     gpio_set_level(pin, 1);  // let pin float, pull up will raise
-    ets_delay_us(11);
+    esp_rom_delay_us (11);
     int r = gpio_get_level(pin);  // Must sample within 15us of start
-    ets_delay_us(48);
+    esp_rom_delay_us (48);
     PORT_EXIT_CRITICAL;
 
     return r;

```

PS: не делаю пулл-реквест, потому что он скорее всего сломает совместимость с `v4`

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

v5.0-beta1

### Which build target have you used?

- [X] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

onewire

### Confirmation

- [X] This report is not a question nor a request for drivers.

trombik, 2022-11-16:
please provide logs, `ets_sys.h: No such file` does not look like the real log to me.

politsin, 2022-11-16:
```
W:/proj/esp32-idf5-ds18b20/components/onewire/onewire.c:55:10: fatal error: ets_sys.h: No such file or directory
   55 | #include <ets_sys.h>
      |          ^~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
```

PR #479 (2022-11-16): bugfix: fix ets_sys.h error
Should fix #477
closed 2022-11-16 by UncleRus