UncleRus/esp-idf-lib #481: error while using ESP-IDF MQTT TCP example

went521, 2022-11-28:
### The issue

As a novice, I encountered the same problem: esp tls: couldn't get hostname for: mqtts. the clouds com: getaddrinfo() returns 202, addrinfo=0x0
How can I solve this problem? If you could help me, I would be very grateful!

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

v4.4.3

### Which build target have you used?

- [X] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

mqtt-tcp

### Anything in the logs that might be useful for us?

```console
esp_mqtt_client_config_t mqtt_cfg = {    
        .host = "mqtts.heclouds.com",
        .client_id=ONENET_clientId,
        .username=ONENET_UserName,
        .password=ONENET_PassWord,
        .port=1883,
    };

E (828) esp-tls: couldn't get hostname for :mqtts.heclouds.com: getaddrinfo() returns 202, addrinfo=0x0
E (838) TRANSPORT_BASE: Failed to open a new connection: 32769
E (848) MQTT_CLIENT: Error transport connect
```

### Confirmation

- [X] This report is not a question nor a request for drivers.

went521, 2022-11-28:
my device is esp-wroom-32

trombik, 2022-11-28:
first, don't lie. this is a question. second, we do not maintain the library. ask at https://www.esp32.com/viewforum.php?f=13

closed 2022-11-28 by trombik