UncleRus/esp-idf-lib #496: sht3x: assert failed: xQueueSemaphoreTake queue.c:1552 (( pxQueue ))

mKainzbauer, 2023-01-22:
### The issue

After moving to idf 5.0.0 I am getting

assert failed: xQueueSemaphoreTake queue.c:1552 (( pxQueue ))

In my code I call

```
sht3x_init_desc(&sht3x_dev, 0, SHT3X_I2C_ADDR_GND, 23, 22)
sht3x_init(&sht3x_dev)
```

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

v5.0.0

### Which build target have you used?

- [X] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

sht3x

### Anything in the logs that might be useful for us?

```console
assert failed: xQueueSemaphoreTake queue.c:1552 (( pxQueue ))

Backtrace: 0x40081b52:0x3ffbafe0 0x4008a0fd:0x3ffbb000 0x40090f81:0x3ffbb020 0x4008af86:0x3ffbb140 0x400e2e3f:0x3ffbb180 0x400e27ae:0x3ffbb1c0 0x400e27cc:0x3ffbb1f0 0x400e2973:0x3ffbb210 0x400d9375:0x3ffbb230 0x400d89cd:0x3ffbb250 0x400d855b:0x3ffbb270 0x4008d479:0x3ffbb290
0x40081b52: panic_abort at C:/Users/user/esp/esp-idf/components/esp_system/panic.c:412

0x4008a0fd: esp_system_abort at C:/Users/user/esp/esp-idf/components/esp_system/esp_system.c:135

0x40090f81: __assert_func at C:/Users/user/esp/esp-idf/components/newlib/assert.c:78

0x4008af86: xQueueSemaphoreTake at C:/Users/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1552 (discriminator 1)

0x400e2e3f: i2c_dev_write at C:/Users/user/esp/humi_temp/components/i2cdev/i2cdev.c:308

0x400e27ae: send_cmd_nolock at C:/Users/user/esp/humi_temp/components/sht3x/sht3x.c:126

0x400e27cc: send_cmd at C:/Users/user/esp/humi_temp/components/sht3x/sht3x.c:132 (discriminator 2)

0x400e2973: sht3x_init at C:/Users/user/esp/humi_temp/components/sht3x/sht3x.c:237 (discriminator 2)
```

### Additional information or context

Didn't happen with esp-idf 4.4

### Confirmation

- [X] This report is not a question nor a request for drivers.

thesparkness, 2023-01-31:
Hello, 
I have the same issue with the bme680 on IDF 5.1 after the init during the first i2c call. It was working fine with the same code with IDF 4.3. 
Thanks for your help

UncleRus, 2023-02-03:
It looks like the error occurs when trying to get an I2C bus mutex. 
Did you initialize the library calling the `i2cdev_init()` function?
Better yet, show an example of source code on which the error is guaranteed to be reproduced.

mKainzbauer, 2023-02-03:
Yes, I do an i2cdev_init();

The following code crashes:

```
#include <string.h>
#include "sht3x.h"
#include "esp_log.h"

static sht3x_t sht3x_dev;

void app_main(void) {
    
    float temperature;
    float humidity;

    memset(&sht3x_dev, 0, sizeof(sht3x_t));
    i2cdev_init();
    sht3x_init_desc(&sht3x_dev, 0, SHT3X_I2C_ADDR_GND, GPIO_NUM_23, GPIO_NUM_22);
    while(true) {
        vTaskDelay(pdMS_TO_TICKS(1000));
        sht3x_init(&sht3x_dev);
        vTaskDelay(pdMS_TO_TICKS(1000));
        sht3x_measure(&sht3x_dev, &temperature, &humidity);
        ESP_LOGD("measure_sht3x", "SHT3x Sensor: %.2f°C, %.2f%%", temperature, humidity);
    }
}
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40086f11  PS      : 0x00060830  A0      : 0x800d59aa  A1      : 0x3ffb4890  
0x40086f11: xQueueSemaphoreTake at C:/Users/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1556

A2      : 0xa5a5a5a5  A3      : 0x00000064  A4      : 0x00060820  A5      : 0x3ffb4960  
A6      : 0x00000064  A7      : 0x000f4240  A8      : 0x8008893e  A9      : 0x3ffb48a0  
A10     : 0x00000003  A11     : 0x00060823  A12     : 0x00060820  A13     : 0x18000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xa5a5a5e5  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x40086f0e:0x3ffb4890 0x400d59a7:0x3ffb48d0 0x400d531a:0x3ffb4910 0x400d5338:0x3ffb4940 0x400d54db:0x3ffb4960 0x400d52e1:0x3ffb4980 0x400e4900:0x3ffb49b0 0x400887c5:0x3ffb49e0
0x40086f0e: xQueueSemaphoreTake at C:/Users/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1552 (discriminator 1)

0x400d59a7: i2c_dev_write at C:/Users/user/esp/sht3x_minimal/components/i2cdev/i2cdev.c:308

0x400d531a: send_cmd_nolock at C:/Users/user/esp/sht3x_minimal/components/sht3x/sht3x.c:126

0x400d5338: send_cmd at C:/Users/user/esp/sht3x_minimal/components/sht3x/sht3x.c:132 (discriminator 2)

0x400d54db: sht3x_init at C:/Users/user/esp/sht3x_minimal/components/sht3x/sht3x.c:237 (discriminator 2)

0x400d52e1: app_main at C:/Users/user/esp/sht3x_minimal/main/sht3x_main.c:18

0x400e4900: main_task at C:/Users/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

0x400887c5: vPortTaskWrapper at C:/Users/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

trombik, 2023-02-09:
what if you enable `CONFIG_FREERTOS_UNICORE` in `sdkconfig`?

mKainzbauer, 2023-02-09:
It is enabled.
Disabling it makes no difference.

mKainzbauer, 2023-02-10:
Solved:

The line:

`sht3x_init_desc(&sht3x_dev, 0, SHT3X_I2C_ADDR_GND, GPIO_NUM_23, GPIO_NUM_22);`

is wrong and should read:

`sht3x_init_desc(&sht3x_dev, SHT3X_I2C_ADDR_GND, 0, GPIO_NUM_23, GPIO_NUM_22);`

Note the 2nd and 3rd parameter (I2C port/Device Address) being exchanged.
I don't know why it worked with idf 4.4, but it worked.

Sorry for bothering everyone, but since this worked before, it was a hard catch for me to find. I found it with downloading a tutorial example which worked and tearing it down line by line.

closed 2023-07-08 by UncleRus