UncleRus/esp-idf-lib #498: TM1621 driver request

kiralikbeyin, 2023-01-29:
### Device name

TM1621

### Device description

New sonoff TH LCD screen

https://templates.blakadder.com/sonoff_THR320D

https://github.com/arendst/Tasmota/issues/15856

### Device product page URL

https://templates.blakadder.com/sonoff_THR320D

### Datasheet URL

https://datasheet.lcsc.com/lcsc/1810191525_TM-Shenzhen-Titan-Micro-Elec-TM1621_C41326.pdf

### Can the device be purchased at any of the followings?

- [X] AliExpress
- [ ] Amazon
- [ ] ebay
- [ ] Digi-Key
- [ ] Mouser

### URLs to purchase the device

https://www.aliexpress.com/item/1005004450591007.html

https://itead.cc/product/sonoff-th-elite-smart-temperature-and-humidity-monitoring-switch/ref/34/

### Other implementations

https://templates.blakadder.com/sonoff_THR320D

https://github.com/arendst/Tasmota/issues/15856

UncleRus, 2023-07-08:
Datasheet in Chinese is a bit too much for me...

UncleRus, 2023-07-08:
OK, here is the code for this display: https://github.com/arendst/Tasmota/blob/d9d9ca2651363ac364f720403657e557e8a0efe8/tasmota/tasmota_xdrv_driver/xdrv_87_esp32_sonoff_tm1621.ino

The bad thing is that specific segments of the display on the Sonoff's device are connected to specific pins of the chip.
The universal driver will consist of a single function that sends an address/command/data to the chip, because there will be no binding of segments to pins - it will remain entirely on the programmer who will use the driver.
Adding hundreds of different chip/LCD configurations to this repository is hell. Thus, I decline your request for a driver for this chip and close this issue.

If you figure out how to write a convenient universal driver, feel free to reopen the issue.

closed 2023-07-08 by UncleRus