UncleRus/esp-idf-lib #523: TSL2591 Light Sensor - return NAN value

UncleRus, 2023-06-02:
### Discussed in https://github.com/UncleRus/esp-idf-lib/discussions/518

<div type='discussions-op-text'>

<sup>Originally posted by **besbar** May 24, 2023</sup>
Dear community, 

Thank you so much for this repository. I've recenty started using ESP-IDF (in order to exploit Bluetooth mesh techno.).
I am using ESP-IDF v5.01.
I wanted to make a BT mesh multi-sensors with so multiple sensors.

Amongst I2C devices, I made working together on a ESP32 Adafruit Feather following sensors:
- SHT31 or AHT20 (°C and %RH),
- MCP9808 (°C)
- SGP30 (TVOC/eCO2),
- OLED Screen SH1107,
- LC709203F (battery monitor),   
- BH1750 or VEML7700 (light sensors)

However I meet difficulties with the TSL2591 light sensor, when I use the component (exactly the same way as for sensors above), results returned are nan Lux.
I am using the TSL2591 ADA1980 from Adafruit.
Is there anybody here that has made this sensor working properly?

It is worth noting that this sensor is working in Arduino IDE with adafruit library.

Thanks a lot for any help on that topic. </div>

UncleRus, 2023-06-02:
@besbar 
You are required to:
1. A short clear example on which the tsl2591 library does not work.
2. Device output log

besbar, 2023-06-04:
Thank you @UncleRus 

**1. A short clear example on which the tsl2591 library does not work.**

I am using **default example** (https://github.com/UncleRus/esp-idf-lib/tree/master/examples/tsl2591/default) in the TSL2591 example dir:

```
#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <esp_system.h>
#include <tsl2591.h>

#ifndef APP_CPU_NUM
#define APP_CPU_NUM PRO_CPU_NUM
#endif

void tsl2591_test(void *pvParameters) {
    tsl2591_t dev = { 0 };

    ESP_ERROR_CHECK(tsl2591_init_desc(&dev, 0, CONFIG_EXAMPLE_I2C_MASTER_SDA, CONFIG_EXAMPLE_I2C_MASTER_SCL));
    ESP_ERROR_CHECK(tsl2591_init(&dev));

    float lux;
    esp_err_t res;
    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(100));

        if ((res = tsl2591_get_lux(&dev, &lux)) != ESP_OK)
            printf("Could not read lux value: %d\n", res);
        else
            printf("Lux: %f\n", lux);
    }
}

void app_main()
{
    ESP_ERROR_CHECK(i2cdev_init());

    xTaskCreatePinnedToCore(tsl2591_test, "tsl2591_test", configMINIMAL_STACK_SIZE * 8, NULL, 5, NULL, APP_CPU_NUM);
}
```

As I use a ESP32 Adafruit Huzzah, SCL and SDA pins are 22 and 23 resp.
So here is the **kconfig.projbuild**

```
menu "Example configuration"

    config EXAMPLE_I2C_MASTER_SCL
        int "SCL GPIO Number"
        default 5 if IDF_TARGET_ESP8266
        default 6 if IDF_TARGET_ESP32C3
        default 22 if IDF_TARGET_ESP32 || IDF_TARGET_ESP32S2 || IDF_TARGET_ESP32S3
        help
            GPIO number for I2C Master clock line.

    config EXAMPLE_I2C_MASTER_SDA
        int "SDA GPIO Number"
        default 4 if IDF_TARGET_ESP8266
        default 5 if IDF_TARGET_ESP32C3
        default 23 if IDF_TARGET_ESP32 || IDF_TARGET_ESP32S2 || IDF_TARGET_ESP32S3
        help
            GPIO number for I2C Master data line.

endmenu
```

I confirm that the main CMakeList.txt is pointing to this repo components folder
```
# The following four lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS ${CMAKE_CURRENT_SOURCE_DIR}/../../../components)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(default)
```

besbar, 2023-06-04:
**2. Device output log**

I (0) cpu_start: App cpu up.
I (210) cpu_start: Pro cpu start user code
I (210) cpu_start: cpu freq: 240000000 Hz
I (210) cpu_start: Application information:
I (215) cpu_start: Project name:     default
I (220) cpu_start: App version:      0.9.2-42-gb867bd5-dirty
I (226) cpu_start: Compile time:     Jun  3 2023 09:38:38
I (232) cpu_start: ELF file SHA256:  9e7eabe81b4928f1...
I (238) cpu_start: ESP-IDF:          v5.0.1-dirty
I (244) cpu_start: Min chip rev:     v0.0
I (248) cpu_start: Max chip rev:     v3.99
I (253) cpu_start: Chip rev:         v3.0
I (258) heap_init: Initializing. RAM available for dynamic allocation:
I (265) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (271) heap_init: At 3FFB2860 len 0002D7A0 (181 KiB): DRAM
I (277) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (284) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (290) heap_init: At 4008CF80 len 00013080 (76 KiB): IRAM
I (298) spi_flash: detected chip: generic
I (301) spi_flash: flash io: dio
I (306) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Lux: nan
Lux: nan
Lux: nan
Lux: nan
Lux: nan

UncleRus, 2023-06-05:
OK, I lost my TSL2591, so I just ordered a new one. It will take about three weeks for it to arrive from China.

UncleRus, 2023-07-08:
I hate Russian Post - they lost my package. Again.
I re-ordered a demo board with this chip, delivered to the postamate, usually such packages arrive without problems.

I'm sorry for the delay.

besbar, 2023-07-11:
Absolutely no worries.

DigitAlexM, 2023-07-12:
Thanks @UncleRus for your commitment.

Even I encountered the same fact of NaN as a result.
I have TSL2591 ADA1980 from Adafruit.
I have tried to research the source of the error but without success, so I am available for any test that may be of help.

Thank you

UncleRus, 2023-07-20:
@juliandoerner Could you look into this problem?

UncleRus, 2023-07-20:
@DigitAlexM @besbar 
The driver shows a lot of debug information, could you turn on the debug logging level in the settings?
(`idf.py menuconfig ->  Component config -> Log output -> Default log verbosity = Debug`)

```
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
CONFIG_LOG_DEFAULT_LEVEL=4
CONFIG_LOG_MAXIMUM_EQUALS_DEFAULT=y
CONFIG_LOG_MAXIMUM_LEVEL=4
```

juliandoerner, 2023-07-20:
Does the fix mentioned in #449 work? I could not reproduce it back then since I don’t not find a spare sensor.

DigitAlexM, 2023-07-20:
Yes of course, i have debugging enabled.
This is the result on the console:
`--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6944
load:0x40078000,len:15500
load:0x40080400,len:3844
0x40080400: _init at ??:?

entry 0x4008064c
I (27) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (27) boot: compile time 00:19:18
I (27) boot: chip revision: v1.0
I (30) boot.esp32: SPI Speed      : 40MHz
I (35) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 2MB
I (44) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0ac9ch ( 44188) map
I (112) esp_image: segment 1: paddr=0001acc4 vaddr=3ffb0000 size=01ff0h (  8176) load
I (115) esp_image: segment 2: paddr=0001ccbc vaddr=40080000 size=0335ch ( 13148) load
I (123) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=19858h (104536) map
I (164) esp_image: segment 4: paddr=00039880 vaddr=4008335c size=09d68h ( 40296) load
I (187) boot: Loaded app from partition at offset 0x10000
I (187) boot: Disabling RNG early entropy source...
I (198) cpu_start: Pro cpu up.
I (199) cpu_start: Starting app cpu, entry point is 0x40081184
0x40081184: call_start_cpu1 at /home/alexmoriconi/ESP-IDF/esp-idf-v5.0.2/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
D (207) clk: RTC_SLOW_CLK calibration value: 3186944
I (218) cpu_start: Pro cpu start user code
I (218) cpu_start: cpu freq: 160000000 Hz
I (218) cpu_start: Application information:
I (223) cpu_start: Project name:     app-template
I (228) cpu_start: App version:      1
I (232) cpu_start: Compile time:     Jul 21 2023 00:19:16
I (239) cpu_start: ELF file SHA256:  54cdb9425a1ab15f...
I (245) cpu_start: ESP-IDF:          v5.0.2-dirty
I (250) cpu_start: Min chip rev:     v0.0
I (255) cpu_start: Max chip rev:     v3.99 
I (259) cpu_start: Chip rev:         v1.0
D (264) memory_layout: Checking 7 reserved memory ranges:
D (270) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (276) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb2938
D (282) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (289) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (295) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (302) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /home/alexmoriconi/ESP-IDF/esp-idf-v5.0.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1743

D (308) memory_layout: Reserved memory range 0x40080000 - 0x4008d0c4
0x40080000: _WindowOverflow4 at /home/alexmoriconi/ESP-IDF/esp-idf-v5.0.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1743

D (315) memory_layout: Building list of available memory regions:
D (321) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (327) memory_layout: Available memory region 0x3ffb2938 - 0x3ffb8000
D (334) memory_layout: Available memory region 0x3ffb8000 - 0x3ffc0000
D (341) memory_layout: Available memory region 0x3ffc0000 - 0x3ffc2000
[... 77 lines cut ...]
D (744) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (754) tsl2591: Red register: 0x0; Data: 0x0.
D (754) tsl2591: Initial enable register: 0.
D (764) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (764) tsl2591: Red register: 0x1; Data: 0x0.
D (774) tsl2591: Initial control register: 0.
D (774) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (784) tsl2591: Red register: 0xc; Data: 0x0.
D (784) tsl2591: Initial persistence filter: 0.
D (1014) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (1014) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (1014) tsl2591: channel0: 0x0 channel1: 0x0.
Lux: nan
D (1124) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (1124) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (1124) tsl2591: channel0: 0x0 channel1: 0x0.
Lux: nan
D (1234) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (1234) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (1234) tsl2591: channel0: 0x0 channel1: 0x0.
Lux: nan
D (1344) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (1344) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (1344) tsl2591: channel0: 0x0 channel1: 0x0.
Lux: nan`

juliandoerner, 2023-07-20:
Given the debug information, the chip is not configured in a working mode. You need to configure the enable and control register. Refer to #449 and data sheet for the correct function calls. Someone with a sensor on hand should fix the example.

UncleRus, 2023-07-21:
Please test #541

DigitAlexM, 2023-07-21:
@juliandoerner
I carried out the test as per reference, I modified the init function with the additional commands.
The result, however, remains the same

`--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6944
load:0x40078000,len:15500
load:0x40080400,len:3844
0x40080400: _init at ??:?

entry 0x4008064c
I (27) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (27) boot: compile time 00:19:18
I (27) boot: chip revision: v1.0
I (30) boot.esp32: SPI Speed      : 40MHz
I (35) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 2MB
I (44) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0ad1ch ( 44316) map
I (112) esp_image: segment 1: paddr=0001ad44 vaddr=3ffb0000 size=01ff0h (  8176) load
I (115) esp_image: segment 2: paddr=0001cd3c vaddr=40080000 size=032dch ( 13020) load
I (123) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=19b8ch (105356) map
I (164) esp_image: segment 4: paddr=00039bb4 vaddr=400832dc size=09de8h ( 40424) load
I (187) boot: Loaded app from partition at offset 0x10000
I (187) boot: Disabling RNG early entropy source...
I (199) cpu_start: Pro cpu up.
I (199) cpu_start: Starting app cpu, entry point is 0x40081184
0x40081184: call_start_cpu1 at /home/alexmoriconi/ESP-IDF/esp-idf-v5.0.2/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
D (207) clk: RTC_SLOW_CLK calibration value: 3183526
I (218) cpu_start: Pro cpu start user code
I (218) cpu_start: cpu freq: 160000000 Hz
I (218) cpu_start: Application information:
I (223) cpu_start: Project name:     app-template
I (229) cpu_start: App version:      1
I (233) cpu_start: Compile time:     Jul 21 2023 00:19:16
I (239) cpu_start: ELF file SHA256:  76357dca0b14806d...
I (245) cpu_start: ESP-IDF:          v5.0.2-dirty
I (250) cpu_start: Min chip rev:     v0.0
I (255) cpu_start: Max chip rev:     v3.99 
I (260) cpu_start: Chip rev:         v1.0
D (265) memory_layout: Checking 7 reserved memory ranges:
D (270) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (277) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb2938
D (283) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (289) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (296) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (302) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /home/alexmoriconi/ESP-IDF/esp-idf-v5.0.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1743

D (309) memory_layout: Reserved memory range 0x40080000 - 0x4008d0c4
0x40080000: _WindowOverflow4 at /home/alexmoriconi/ESP-IDF/esp-idf-v5.0.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1743

D (315) memory_layout: Building list of available memory regions:
D (321) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (328) memory_layout: Available memory region 0x3ffb2938 - 0x3ffb8000
[... 75 lines cut ...]
D (724) tsl2591: Initialize sensor.
E (1744) i2cdev: [0x29 at 0] Could not take device mutex
E (2754) i2cdev: [0x29 at 0] Could not take device mutex
E (3764) i2cdev: [0x29 at 0] Could not take device mutex
E (4774) i2cdev: [0x29 at 0] Could not take device mutex
D (4774) i2cdev: Reconfiguring I2C driver on port 0
D (4774) intr_alloc: Connected src 49 to int 3 (cpu 1)
D (4774) i2cdev: I2C driver successfully reconfigured on port 0
D (4784) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (4784) tsl2591: Red register: 0x0; Data: 0x0.
D (4794) tsl2591: Initial enable register: 0.
D (4794) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (4804) tsl2591: Red register: 0x1; Data: 0x0.
D (4804) tsl2591: Initial control register: 0.
D (4804) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (4814) tsl2591: Red register: 0xc; Data: 0x0.
D (4814) tsl2591: Initial persistence filter: 0.
D (5054) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (5054) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (5054) tsl2591: channel0: 0x0 channel1: 0x0.
Lux: nan
D (5164) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (5164) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (5164) tsl2591: channel0: 0x0 channel1: 0x0.
Lux: nan`

UncleRus, 2023-07-21:
@DigitAlexM Please do not add function calls that set modes of operation directly to driver initialization - this is bad practice, as what happens is implicit.

It is better to call these functions yourself and set the device operation mode you need.
I ask you to test the changes from the PR #541.

DigitAlexM, 2023-07-21:
@UncleRus 
Ok, I followed the #541 and now it doesn't seem to come back NaN.

`--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6944
load:0x40078000,len:15500
load:0x40080400,len:3844
0x40080400: _init at ??:?

entry 0x4008064c
I (27) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (27) boot: compile time 11:11:58
I (27) boot: chip revision: v1.0
I (30) boot.esp32: SPI Speed      : 40MHz
I (35) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 2MB
I (44) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0a8fch ( 43260) map
I (111) esp_image: segment 1: paddr=0001a924 vaddr=3ffb0000 size=01f78h (  8056) load
I (115) esp_image: segment 2: paddr=0001c8a4 vaddr=40080000 size=03774h ( 14196) load
I (123) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=19604h (103940) map
I (163) esp_image: segment 4: paddr=0003962c vaddr=40083774 size=098bch ( 39100) load
I (186) boot: Loaded app from partition at offset 0x10000
I (186) boot: Disabling RNG early entropy source...
I (197) cpu_start: Pro cpu up.
I (198) cpu_start: Starting app cpu, entry point is 0x40081170
0x40081170: call_start_cpu1 at /home/alexmoriconi/ESP-IDF/esp-idf-v5.0.2/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
D (206) clk: RTC_SLOW_CLK calibration value: 3185702
I (217) cpu_start: Pro cpu start user code
I (217) cpu_start: cpu freq: 160000000 Hz
I (217) cpu_start: Application information:
I (222) cpu_start: Project name:     app-template
I (227) cpu_start: App version:      1
I (231) cpu_start: Compile time:     Jul 21 2023 11:11:57
I (238) cpu_start: ELF file SHA256:  ba1b415aaea52984...
I (244) cpu_start: ESP-IDF:          v5.0.2-dirty
I (249) cpu_start: Min chip rev:     v0.0
I (254) cpu_start: Max chip rev:     v3.99 
I (258) cpu_start: Chip rev:         v1.0
D (263) memory_layout: Checking 7 reserved memory ranges:
D (269) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (275) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb28c0
D (281) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (288) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (294) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (301) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /home/alexmoriconi/ESP-IDF/esp-idf-v5.0.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1743

D (307) memory_layout: Reserved memory range 0x40080000 - 0x4008d030
0x40080000: _WindowOverflow4 at /home/alexmoriconi/ESP-IDF/esp-idf-v5.0.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1743

D (314) memory_layout: Building list of available memory regions:
D (320) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (326) memory_layout: Available memory region 0x3ffb28c0 - 0x3ffb8000
D (333) memory_layout: Available memory region 0x3ffb8000 - 0x3ffc0000
[... 98 lines cut ...]
Lux: 7.858998
D (2163) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (2163) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (2163) tsl2591: channel0: 0x13c channel1: 0x67.
Lux: 7.810359
D (2673) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (2673) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (2673) tsl2591: channel0: 0x13c channel1: 0x67.
Lux: 7.810359
D (3183) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (3183) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (3183) tsl2591: channel0: 0x13d channel1: 0x67.
Lux: 7.858998
D (3693) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (3693) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (3693) tsl2591: channel0: 0x140 channel1: 0x68.
Lux: 7.931520
D (4203) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (4203) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (4203) tsl2591: channel0: 0x140 channel1: 0x69.
Lux: 7.858250
D (4713) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (4713) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (4713) tsl2591: channel0: 0x141 channel1: 0x69.
Lux: 7.906811`

UncleRus, 2023-07-21:
Cool! Thank you!

PR #541 (2023-07-21): fix: add sensor init to default tsl2591 example
Fixes #523
closed 2023-07-21 by UncleRus