UncleRus/esp-idf-lib #525: Labeler broken

UncleRus, 2023-06-02:
### The issue

```
Error: HttpError: Bad credentials
Error: Bad credentials
```
See https://github.com/UncleRus/esp-idf-lib/actions/runs/5157823415/jobs/9290732206?pr=524#step:2:7

### Confirmation

- [X] This report is not a bug report in code, a question, nor a request for drivers.

trombik, 2023-06-03:
I cannot think of any issues in the GitHub action. it's been working, the version of the GitHub Action is the latest, and no breaking changes in recent GitHub development. maybe, the GitHub token has expired?

UncleRus, 2023-06-03:
I completely forgot how the token is generated and where exactly it is located :-)
Сould you remind me?

trombik, 2023-06-03:
It's probably [Developer settings](https://github.com/settings/apps).

https://docs.github.com/en/authentication/keeping-your-account-and-data-secure/managing-your-personal-access-tokens

UncleRus, 2023-06-03:
Yes, the token was the problem - it expired on Jun 1.
I regenerated it by changing the expiration date.
However, restarting the failed job after token regeneration did nothing: the same error. It looks like something needs to be changed in the CI scripts after token regeneration:

[image]

trombik, 2023-06-06:
as the tests have been working, it probably was caused by some delay, i guess.