UncleRus/esp-idf-lib #531: Support for Sensirion STS3X

slimcdk, 2023-07-03:
### Device name

STS3X

### Device description

Device is a digital temperature sensor. It communicates using I2C.

### Device product page URL

https://www.sensirion.com/search/products?q=STS3x

### Datasheet URL

https://www.sensirion.com/media/documents/1C706A20/6164200F/Sensirion_Temperature_Sensors_STS3xA_Datasheet.pdf

### Can the device be purchased at any of the followings?

- [ ] AliExpress
- [ ] Amazon
- [ ] ebay
- [ ] Digi-Key
- [X] Mouser

### URLs to purchase the device

https://eu.mouser.com/c/sensors/temperature-sensors/board-mount-temperature-sensors/?m=Sensirion&series=STS3x-DIS

slimcdk, 2023-07-03:
Pull request opened #532

UncleRus, 2023-07-05:
Hmm... Did you check your PR on real device?
The datasheet says that the raw data size is 3 bytes. And in your code, the raw data size is 6 bytes.

[image]

```C
#define STS3X_RAW_DATA_SIZE 6

typedef uint8_t sts3x_raw_data_t[STS3X_RAW_DATA_SIZE];
```

UncleRus, 2023-07-05:
I'm rolling back the merge of your PR until I get confirmation that the driver works with the actual device.

slimcdk, 2023-07-11:
Sorry for the radio silence! Still have a bit of vacation left, so I'll explain what I have currently tested. You are right, that the size is only 3 bytes due to the lack of humidity compared to the SHT (which this implementation is an exact copy off). I have tested the driver with an ESP32-S3 and STS31 at 100 MHz and didn't encounter any issues for the week I have been using it.

I'll adjust the size allocation next week and test before submitting a new PR.

slimcdk, 2023-07-19:
@UncleRus I have verified that the smaller allocation works.

slimcdk, 2023-07-19:
Had some conflicts with the files were added and then removed from upstream, so I couldn't figure out to sync properly, so I made a new branch.

Driver is tested on real hardware and is reporting correct measurements.

closed 2023-07-20 by UncleRus