UncleRus/esp-idf-lib #538: CI broken for esp-idv v4.x and esp8266

UncleRus, 2023-07-19:
### The issue

Looks like incompatibility with Python version...

See https://github.com/UncleRus/esp-idf-lib/actions/runs/5598336404/jobs/10237858788

### Confirmation

- [X] This report is not a bug report in code, a question, nor a request for drivers.

UncleRus, 2023-07-20:
I don't understand why in one case pip downloads the wheel package and in the other the source package...
Properly working check:
```
Collecting gevent<2.0,>=1.2.2 (from gdbgui==0.13.2.0->-r /home/runner/work/esp-idf-lib/esp-idf-lib/idf/requirements.txt (line 18))
  Downloading https://dl.espressif.com/pypi/gevent/gevent-1.5.0-cp310-cp310-linux_x86_64.whl (5.7 MB)
```
Malfunctioning check:
```
Collecting gevent<2.0,>=1.2.2 (from gdbgui==0.13.2.0->-r /home/runner/work/esp-idf-lib/esp-idf-lib/idf/requirements.txt (line 17))
  Downloading gevent-1.5.0.tar.gz (5.3 MB)
```

UncleRus, 2023-07-20:
I will temporarily disable checks for esp-idf version 4.2.3 as a workaround and leave the issue open.

trombik, 2023-07-20:
i'm facing the same issue in a different project: `cython`, which released a major version, 3.0 a few days ago, is known to be seriously broken. will see if a workaround works in the CI.

closed 2023-08-05 by UncleRus