UncleRus/esp-idf-lib #539: DS3231 + OLED SSD1306 is not working

DerPicknicker, 2023-07-20:
### The issue

Hi, 

first I have to thank you for the great documentation and the structure of the Git... My Problem seems not that related to IDF-Libs but I guess the issue is caused by DS3231 Lib.. 

Here is my Setup: 

I use an SSD1306 in I2C-HW 1 and the DS3231 in I2C-HW 0. I want to separate them. In both cases the ESP32-S3 is the master obvious because how should the OLED Display and the RTC get the Data? 

The OLED Library is from here: https://github.com/nopnop2002/esp-idf-ssd1306 

I can easily switch the OLED to the other I2C Port, but if I try to combine both components the Code crashes. 

Please have a look. 

Here is the Crash Log: 0x40375781: esp_restart_noos_dig at /Users/USERNAME/esp/esp-idf/components/esp_system/esp_system.c:46 (discriminator 1)

Here is the Code I use - I comment out a Lot just for debugging purpose. If I disable the SetTime Function the Code runs without crashing. So the Write-Function seems to be buggy? 

```
#include <stdio.h>

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"

#include "ssd1306.h"
#include "font8x8_basic.h"
#include <esp_err.h>
#include "../components/esp_idf_lib_helpers/esp_idf_lib_helpers.h"
#include "../components/ds3231/ds3231.h"

SSD1306_t dev;
int center, top, bottom;
char lineChar[20];
#define tag "SSD1306"

void OLED_INIT(){
   	
	
#if CONFIG_I2C_INTERFACE
	ESP_LOGI(tag, "INTERFACE is i2c");
	ESP_LOGI(tag, "CONFIG_SDA_GPIO=%d",CONFIG_SDA_GPIO);
	ESP_LOGI(tag, "CONFIG_SCL_GPIO=%d",CONFIG_SCL_GPIO);
	ESP_LOGI(tag, "CONFIG_RESET_GPIO=%d",CONFIG_RESET_GPIO);
	i2c_master_init(&dev, CONFIG_SDA_GPIO, CONFIG_SCL_GPIO, CONFIG_RESET_GPIO);
#endif // CONFIG_I2C_INTERFACE

#if CONFIG_SPI_INTERFACE
	ESP_LOGI(tag, "INTERFACE is SPI");
	ESP_LOGI(tag, "CONFIG_MOSI_GPIO=%d",CONFIG_MOSI_GPIO);
	ESP_LOGI(tag, "CONFIG_SCLK_GPIO=%d",CONFIG_SCLK_GPIO);
	ESP_LOGI(tag, "CONFIG_CS_GPIO=%d",CONFIG_CS_GPIO);
	ESP_LOGI(tag, "CONFIG_DC_GPIO=%d",CONFIG_DC_GPIO);
	ESP_LOGI(tag, "CONFIG_RESET_GPIO=%d",CONFIG_RESET_GPIO);
	spi_master_init(&dev, CONFIG_MOSI_GPIO, CONFIG_SCLK_GPIO, CONFIG_CS_GPIO, CONFIG_DC_GPIO, CONFIG_RESET_GPIO);
#endif // CONFIG_SPI_INTERFACE

#if CONFIG_FLIP
	dev._flip = true;
	ESP_LOGW(tag, "Flip upside down");
#endif

#if CONFIG_SSD1306_128x64
	ESP_LOGI(tag, "Panel is 128x64");
	ssd1306_init(&dev, 128, 64);
#endif // CONFIG_SSD1306_128x64
#if CONFIG_SSD1306_128x32
	ESP_LOGI(tag, "Panel is 128x32");
	ssd1306_init(&dev, 128, 32);
#endif // CONFIG_SSD1306_128x32
}

void app_main(void)
{
    OLED_INIT();
    ssd1306_clear_screen(&dev, false);
	ssd1306_contrast(&dev, 0xff);
  	ssd1306_display_text_x3(&dev, 0, "Hello", 5, false);
  	vTaskDelay(3000 / portTICK_PERIOD_MS);
	ssd1306_clear_screen(&dev, false);
	ssd1306_contrast(&dev, 0xff);
	char buffTime[8];
	i2c_dev_t devRTC;
    memset(&devRTC, 0, sizeof(i2c_dev_t));

    ESP_ERROR_CHECK(ds3231_init_desc(&devRTC, 1, 11, 12));

    // setup datetime: 2016-10-09 13:50:10
    struct tm time = {
        .tm_year = 223, //since 1900 (2016 - 1900)
        .tm_mon  = 7,  // 0-based
        .tm_mday = 19,
        .tm_hour = 12,
        .tm_min  = 53,
        .tm_sec  = 0
    };
    ESP_ERROR_CHECK(ds3231_set_time(&devRTC, &time));

    

    float temp;

    vTaskDelay(pdMS_TO_TICKS(250));

        

   /* if (ds3231_get_time(&devRTC, &time) != ESP_OK)
     {
        printf("Could not get time\n");
      //  continue;
    }*/

        /* float is used in printf(). you need non-default configuration in
         * sdkconfig for ESP8266, which is enabled by default for this
         * example. see sdkconfig.defaults.esp8266
         */
	sprintf(buffTime,"%02d:%02d",time.tm_hour, time.tm_min);
    //ssd1306_display_text_x3(&dev,0,buffTime,strlen(buffTime),0);
		//printf("%04d-%02d-%02d %02d:%02d:%02d, %.2f deg Cel\n", time.tm_year + 1900 /*Add 1900 for better readability*/, time.tm_mon + 1,
         //   time.tm_mday, time.tm_hour, time.tm_min, time.tm_sec, temp);
     

}

```

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

ESP-IDF v5.05.

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [X] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

DS3231

### Confirmation

- [X] This report is not a question nor a request for drivers.

UncleRus, 2023-07-20:
You cannot use i2cdev based libraries on the same I2C port with non-i2cdev-based libraries.
So either use devices on different I2C ports, or port the ssd1306 library to i2cdev.

closed 2023-07-20 by UncleRus

DerPicknicker, 2023-07-20:
@UncleRus ... I tested both ways.. It still doesn't work. I used port1 for display and port 0 for the Rtc and inversed. So here I guess the issue must be something different.