UncleRus/esp-idf-lib #549: CI: Upload components to Espressif registry

UncleRus, 2023-08-05:
### The issue

We need to upload components to the Expressif registry.

I propose to create a separate workflow, launched manually, for the complete uploading of all components at the same time.

We also need to add a job to upload only changed components into the existing `ci.yml`.

In this regard, it is necessary to provide all components with Espressif metadata (see, for example, https://github.com/trombik/eil_helpers/blob/main/components/eil_helpers/idf_component.yml). We will probably be able to generate this metadata using devtool.

### Confirmation

- [X] This report is not a bug report in code, a question, nor a request for drivers.

UncleRus, 2023-08-07:
Individual components versioning added in #554 
Generation of component manfiests added in #555

nopnop2002, 2025-02-09:
Thank you for publishing such a useful library.

I recently discovered that we can clone only part of a component.

Create the following idf_component.yml in the same hierarchy as main.c.

```
dependencies:
  UncleRus/esp_idf_lib_helpers:
    path: components/esp_idf_lib_helpers/
    git: https://github.com/UncleRus/esp-idf-lib.git
  UncleRus/hd44780:
    path: components/hd44780/
    git: https://github.com/UncleRus/esp-idf-lib.git
```

When we build a projects esp-idf will automaticly fetch repository to managed_components dir and link with your code.

```
$ ls managed_components/
UncleRus__esp_idf_lib_helpers  UncleRus__hd44780
```

trombik, 2025-08-01:
resolved by https://github.com/esp-idf-lib/core