UncleRus/esp-idf-lib #577: mhz19b build error on esp32c6

AxelLin, 2023-09-27:
### The issue

Build on esp-idf-v5.1 branch, with **idf.py set-target esp32c6**.

Got below build error:
esp-idf-lib/components/mhz19b/mhz19b.c: In function 'mhz19b_init':
/home/axel/esp/esp-idf-dev/esp-idf-lib/components/mhz19b/mhz19b.c:64:23: error: 'UART_SCLK_APB' undeclared (first use in this function); did you mean 'UART_SCLK_RTC'?
   64 |         .source_clk = UART_SCLK_APB,
      |                       ^~~~~~~~~~~~~
      |                       UART_SCLK_RTC
/home/axel/esp/esp-idf-dev/esp-idf-lib/components/mhz19b/mhz19b.c:64:23: note: each undeclared identifier is reported only once for each function it appears in
[1055/1357] Building C object esp-idf/noise/CMakeFiles/__idf_noise.dir/noise.c.objninja: build stopped: subcommand failed.

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

v5.1.1-439-gcb174b0fe1

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [X] other

### Component causing the issue

mhz19b

### Confirmation

- [X] This report is not a question nor a request for drivers.

AxelLin, 2023-09-27:
https://github.com/espressif/esp-idf/issues/12314#issuecomment-1736941800
UART_SCLK_APB can be changed to UART_SCLK_DEFAULT.

closed 2023-09-27 by UncleRus

AxelLin, 2023-10-25:
Hi @UncleRus 
I just notice esp32c6 is not official supported? (Above target build does not include esp32c6)
Just wondering if you consider to add esp32c6 support since it is available in esp-idf5.1+.

UncleRus, 2023-10-26:
Strange. We already support `UART_SCLK_DEFAULT`

```C
    ...
    uart_config_t uart_config = {
        .baud_rate = 9600,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 0, 0)
        .source_clk = UART_SCLK_DEFAULT,
#elif ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 3, 0)
        .source_clk = UART_SCLK_APB,
#endif
    };
    ...
```

AxelLin, 2023-10-26:
Hi @UncleRus 

oops, I might make you misunderstand ...
It works for me, however I notice in https://github.com/UncleRus/esp-idf-lib page
The "Supported on" filed for all components does not include support for esp32c6.