UncleRus/esp-idf-lib #579: SR04 ultrasonic sonar issue

weiwi, 2023-09-29:
### The issue

I am using ESP32S3, and kept getting when trying to get sonar readings. I have tried both SR04 and AJ-SR04M sonar boards and have similar results:

1. Sonar heads are sending signals (LED blinks on SR04M, and I can hear both heads sending signals from noise cancelling headphone)
2. Both will get error: Error 513: Ping timeout (no device found)

Can anyone shed some light on this?

Thanks

W

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

master

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [X] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

ultrasonic

### Confirmation

- [X] This report is not a question nor a request for drivers.

hayschan, 2024-03-15:
I recommend trying your code on a simulator, such as the [Wokwi Simulator](https://wokwi.com/). This approach allows you to eliminate potential wiring and hardware issues by simulating your setup. 

Wokwi offers an [HC-SR04 ultrasonic sensor simulator](https://docs.wokwi.com/parts/wokwi-hc-sr04) which might be relevant for your needs. 

Testing your code in a controlled environment like this can help you identify whether the issue is with the code, wiring, or the hardware components themselves.

Tarungtm, 2024-04-24:
Hi @UncleRus,
I am using ESP32 wrover ie and esp32 ce-3 mini. I have tried to fetch the distance using SR04M-2 but facing errors like.
1. ping timeout(no device found).
2. echo timeout (i.e. distance too big).

I tried to debug the program came to know that it is not fetching timer value in "ultrasonic_meaure_raw" function.

Can you help me with it.

trombik, 2025-08-01:
moved to: https://github.com/esp-idf-lib/ultrasonic/issues/2

closed 2025-08-01 by trombik