UncleRus/esp-idf-lib #582: ACK issue in wake up command in SCD4x component

vbieleny, 2023-10-20:
### The issue

I have tried to connect my SCD41 and I uploaded the [example](https://github.com/UncleRus/esp-idf-lib/blob/master/examples/scd4x/default/main/main.c) code to my [ESP32-C3-DevKitM-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html) and I got this error:
```
E (282) i2cdev: Could not write to device [0x62 at 0]: -1 (ESP_FAIL)
```

After some troubleshooting, I found out that the issue is in `scd4x_wake_up` function. So I checked the [datasheet](https://sensirion.com/media/documents/48C4B7FB/64C134E7/Sensirion_SCD4x_Datasheet.pdf) and section 3.10.4 has a description where it is mentioned that this command is not acknowledged by SCD4x:
> Wake up the sensor from sleep mode into idle mode. Note that the SCD4x does not acknowledge the wake_up
command. The sensor idle state after wake up can be verified by reading out the serial number (Section 3.9.2).

I've checked the code for the wake up function and it's using `i2c_dev_write` function, which always enables ACK, which I think is why this function fails.

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

v5.1.1

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [X] other

### Component causing the issue

scd4x

### Confirmation

- [X] This report is not a question nor a request for drivers.

AxelLin, 2023-11-29:
> ### The issue
> I have tried to connect my SCD41 and I uploaded the [example](https://github.com/UncleRus/esp-idf-lib/blob/master/examples/scd4x/default/main/main.c) code to my [...]

I can confirm the same error, I have to remove ESP_ERROR_CHECK around scd4x_wake_up() to make it work.

nup002, 2024-11-12:
Wasted a couple of hours on this.

trombik, 2025-02-27:
Would you create a PR for this issue?

vbieleny, 2025-04-14:
> Would you create a PR for this issue?

@trombik I can try when I'll have the time for it.

trianglesis, 2025-04-19:
Hello,
I'm seeing a similar issue with SCD41.

IDF 5.4.1
lib: master

```log
I (10157) CO2 SCD41 INIT: run scd4x_get_serial_number
D (10167) i2cdev: Reconfiguring I2C driver on port 0
D (10167) intr_alloc: Connected src 50 to int 15 (cpu 0)
D (10177) i2cdev: I2C driver successfully reconfigured on port 0
D (10187) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (10177) wifi:mms: 0->0
E (10187) i2cdev: Could not write to device [0x62 at 0]: -1 (ESP_FAIL)
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x42011308
--- 0x42011308: sensor_init at D:/Projects/ESP/projects/ESP32-C6-OLED/Air_Quality_station/main/sensor_co2/co2_sensor.c:164 (discriminator 1)

file: "./main/sensor_co2/co2_sensor.c" line 164
func: sensor_init
expression: scd4x_get_serial_number(&dev, serial, serial + 1, serial + 2)

abort() was called at PC 0x4080b6d1 on core 0
--- 0x4080b6d1: _esp_error_check_failed at D:/Projects/ESP/Espressif/v5.4.1/esp-idf/components/esp_system/esp_err.c:49
```

I've read datasheets for this sensor and tried to switch two of them, one working on an ESPHome device.

I have a complete run here, there are a lot of words...

[README.md](https://github.com/trianglesis/Air_Quality_station/blob/4802c416501cf49720c3299119afd54bb335f894/doc/how_to_basics/README.md#scd4x)

trombik, 2025-08-01:
moved to: https://github.com/esp-idf-lib/i2cdev/issues/4 https://github.com/esp-idf-lib/scd4x/issues/2

closed 2025-08-01 by trombik