UncleRus/esp-idf-lib #586: cmake fails

omani, 2023-12-10:
### The issue

`cmake ..` in `./build` directory fails after editing `CMakeLists.txt`.

```
CMake Error at
  esp/esp-idf-lib/components/color/CMakeLists.txt:1 (if):

    if given arguments:

      "STREQUAL" "4" "AND" "STREQUAL" "1" "AND" "STREQUAL" "3"

    Unknown arguments specified
```

I get this error for all components it seems.

afaic it is because it expects esp-idf rather than what I am using (esp8266 rtos sdk). the rtos sdk has no variable `IDF_VERSION_MAJOR` and `IDF_VERSION_MINOR`. but Im not sure if it is because of this, since I am new to cmake.

### Which SDK are you using?

ESP8266 RTOS SDK

### Which version of SDK are you using?

master

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [X] esp8266
- [ ] other

### Component causing the issue

hx711

### Confirmation

- [X] This report is not a question nor a request for drivers.

omani, 2023-12-10:
I see these variables do exist in the ESP8266 RTOS SDK. they are defined in:

```
~/esp/sdk/ESP8266_RTOS_SDK/tools/cmake$ rg IDF_VERSION
version.cmake
1:set(IDF_VERSION_MAJOR 4)
2:set(IDF_VERSION_MINOR 0)
3:set(IDF_VERSION_PATCH 0)
```

but including version.cmake in my CMakeLists.txt does not solve the problem. neither works a different order of the directives in the CMakeLists.txt.

maybe someone can help me out here. as far as I understand, this whole esp-idf-lib repo is not usable with the esp8266 sdk when it comes to cmake. all components have the IDF_VERSION_* variable check in them. I wonder why nobody came up with this error until now.

omani, 2023-12-10:
maybe related #276

trombik, 2023-12-17:
when I ported the libraries to `esp8266`, the SDK was, and probably still is, unusable with `idf.py`, i.e. `cmake`-based build. the SDK has been abandoned (see the issues and the PRs in the repo). I do not follow the development of the SDK anymore. that is why only `make` build is in the CI.