UncleRus/esp-idf-lib #590: MCP23017 Example throws an error when using interrupts

sazanof, 2024-01-01:
### The issue

```
ESP-IDF v5.1.2 2nd stage bootloader
Multicore bootloader
chip revision: v1.0
SPI Speed      : 40MHz
SPI Mode       : DIO
SPI Flash Size : 2MB
```

Hello! Iuse ESP-WROOM-32 with mcp23017.
Base scnhematics is like this
[image]

SCL - GPIO22
SDA - GPIO21
INTA - GPIO26.

Code - the same as in your example.

After click button on 0 pin, esp23 thrown an error

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

5.1.2

### Which build target have you used?

- [X] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

mcp23x17

### Anything in the logs that might be useful for us?

```console
Error after button click

abort() was called at PC 0x40082cab on core 0
0x40082cab: esp_intr_noniram_disable at /home/sazanof/esp/esp-idf/components/esp_hw_support/intr_alloc.c:855

Backtrace: 0x4008183e:0x3ffb0b80 0x40086b5d:0x3ffb0ba0 0x4008c282:0x3ffb0bc0 0x40082cab:0x3ffb0c30 0x40082de9:0x3ffb0c60 0x40082e62:0x3ffb0c80 0x400dbff6:0x3ffb0cb0 0x400dc043:0x3ffb0cf0 0x40083272:0x3ffb0d10 0x40083352:0x3ffb0d30 0x400833e2:0x3ffb0d60 0x40082a9d:0x3ffb0d90 0x40084dab:0x3ffb5320 0x400d209f:0x3ffb5340 0x40087cba:0x3ffb5360 0x4008920d:0x3ffb5380
0x4008183e: call_start_cpu0 at /home/sazanof/esp/esp-idf/components/esp_system/port/cpu_start.c:415 (discriminator 3)

0x40086b5d: check_trans_valid at /home/sazanof/esp/esp-idf/components/driver/spi/gpspi/spi_master.c:774 (discriminator 5)

0x4008c282: xQueueReceive at /home/sazanof/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1484 (discriminator 1)

0x40082cab: esp_intr_noniram_disable at /home/sazanof/esp/esp-idf/components/esp_hw_support/intr_alloc.c:855

0x40082de9: esp_intr_enable at /home/sazanof/esp/esp-idf/components/esp_hw_support/intr_alloc.c:802

0x40082e62: esp_intr_disable at /home/sazanof/esp/esp-idf/components/esp_hw_support/intr_alloc.c:821 (discriminator 1)

0x400dbff6: _onewire_write_bit at /home/sazanof/esp/eth/esp-idf-lib/components/onewire/onewire.c:143

0x400dc043: _onewire_read_bit at /home/sazanof/esp/eth/esp-idf-lib/components/onewire/onewire.c:163

0x40083272: _xt_syscall_exc at /home/sazanof/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:896

0x40083352: _xt_coproc_exc at /home/sazanof/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1116

0x400833e2: _xt_coproc_exc at /home/sazanof/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1143

0x40082a9d: trace_malloc at /home/sazanof/esp/esp-idf/components/heap/include/heap_trace.inc:97

0x40084dab: bootloader_flash_gpio_config at /home/sazanof/esp/esp-idf/components/bootloader_support/bootloader_flash/src/bootloader_flash_config_esp32.c:96

0x400d209f: _stext at ??:?

0x40087cba: tsens_code_read at /home/lxf/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:1480

0x4008920d: set_chan_dig_gain at /home/lxf/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:1952
```

sometimes an error thrown like this

```
abort() was called at PC 0x40082cab on core 0
0x40082cab: lock_acquire_generic at /home/sazanof/esp/esp-idf/components/newlib/locks.c:130

Backtrace: 0x4008183e:0x3ffb0b80 0x40086b5d:0x3ffb0ba0 0x4008c282:0x3ffb0bc0 0x40082cab:0x3ffb0c30 0x40082de9:0x3ffb0c60 0x40082e62:0x3ffb0c80 0x400dbff6:0x3ffb0cb0 0x400dc043:0x3ffb0cf0 0x40083272:0x3ffb0d10 0x40083352:0x3ffb0d30 0x400833e2:0x3ffb0d60 0x40082a9d:0x3ffb0d90 0x4000bfed:0x3ffb7d40 0x400894f3:0x3ffb7d50 0x40087421:0x3ffb7d70 0x400d83ce:0x3ffb7db0 0x400d5aee:0x3ffb7df0 0x400d5c8f:0x3ffb7e30 0x400d5d64:0x3ffb7e60 0x400d5e49:0x3ffb7e90 0x400d5824:0x3ffb7eb0 0x4008920d:0x3ffb7f00
0x4008183e: panic_abort at /home/sazanof/esp/esp-idf/components/esp_system/panic.c:452

0x40086b5d: esp_system_abort at /home/sazanof/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x4008c282: abort at /home/sazanof/esp/esp-idf/components/newlib/abort.c:38

0x40082cab: lock_acquire_generic at /home/sazanof/esp/esp-idf/components/newlib/locks.c:130

0x40082de9: _lock_acquire_recursive at /home/sazanof/esp/esp-idf/components/newlib/locks.c:158

0x40082e62: __retarget_lock_acquire_recursive at /home/sazanof/esp/esp-idf/components/newlib/locks.c:314

0x400dbff6: _puts_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/puts.c:89 (discriminator 2)

0x400dc043: puts at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/puts.c:129

0x40083272: intr_handler at /home/sazanof/esp/eth/esp-idf-lib/examples/mcp23x17/mcp23017/main/main.c:10

0x40083352: gpio_isr_loop at /home/sazanof/esp/esp-idf/components/driver/gpio/gpio.c:463

0x400833e2: gpio_intr_service at /home/sazanof/esp/esp-idf/components/driver/gpio/gpio.c:485

0x40082a9d: _xt_lowint1 at /home/sazanof/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240

0x4000bfed: _xtos_set_intlevel in ROM

0x400894f3: vPortClearInterruptMaskFromISR at /home/sazanof/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:568
 (inlined by) vPortExitCritical at /home/sazanof/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:532

0x40087421: xQueueReceive at /home/sazanof/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1576

0x400d83ce: i2c_master_cmd_begin at /home/sazanof/esp/esp-idf/components/driver/i2c/i2c.c:1557

0x400d5aee: i2c_dev_read at /home/sazanof/esp/eth/esp-idf-lib/components/i2cdev/i2cdev.c:295

0x400d5c8f: i2c_dev_read_reg at /home/sazanof/esp/eth/esp-idf-lib/components/i2cdev/i2cdev.c:334

0x400d5d64: write_reg_bit_16 at /home/sazanof/esp/eth/esp-idf-lib/components/mcp23x17/mcp23x17.c:125 (discriminator 2)

0x400d5e49: mcp23x17_set_level at /home/sazanof/esp/eth/esp-idf-lib/components/mcp23x17/mcp23x17.c:448

0x400d5824: test at /home/sazanof/esp/eth/esp-idf-lib/examples/mcp23x17/mcp23017/main/main.c:36 (discriminator 1)

0x4008920d: vPortTaskWrapper at /home/sazanof/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

Can you tell me what this might be related to? Is this really a bug? Thanks!
```

### Confirmation

- [X] This report is not a question nor a request for drivers.

UncleRus, 2024-01-13:
Yes, in the example there is a bug in the interrupt handler. 
You cannot call `printf()` directly from an interrupt handler. Instead, you need to use either the FreeRTOS queue or event groups. You can see how to do this correctly in the example for tca95x5.

I'm leaving this issue open until I fix the bug in the mcp23017 example.

sazanof, 2024-01-16:
Thank you. I think, my question is closed. I use code like this

```
static void IRAM_ATTR intr_handler(void *arg)
{

    if (!interrupt_running)
    {
        interrupt_running = true;
        xTaskCreate(read_interrupt, "read_interrupt", configMINIMAL_STACK_SIZE * 6, NULL, 5, NULL);
    }
}
```

and if I need to print - I use ESP_DRAM_LOGx functions

PR #600 (2024-02-03): fix: fix example for mcp23017
Fixes #590
closed 2024-02-03 by UncleRus