UncleRus/esp-idf-lib #598: SHT4x often gives "i2cdev: Could not read from device [0x44 at 0]" error on measurement

MRusinov, 2024-02-02:
### The issue

SHT4x often gives "i2cdev: Could not read from device [0x44 at 0]" error on measurement. ESP32C3

Corrected this way:
```Diff
--- sht4x.c.old	2024-02-02 10:34:14.680822283 +0400
+++ sht4x.c	2024-02-02 11:23:58.529277599 +0400
@@ -174,8 +174,8 @@
 {
     I2C_DEV_TAKE_MUTEX(&dev->i2c_dev);
     I2C_DEV_CHECK(&dev->i2c_dev, send_cmd_nolock(dev, cmd));
-    if (delay_ticks)
-        vTaskDelay(delay_ticks);
+    // if (delay_ticks)
+    vTaskDelay(delay_ticks + 1);
     I2C_DEV_CHECK(&dev->i2c_dev, read_res_nolock(dev, res));
     I2C_DEV_GIVE_MUTEX(&dev->i2c_dev);
```

After that all errors gone...

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

5.1.2

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [X] other

### Component causing the issue

sht4x

### Confirmation

- [X] This report is not a question nor a request for drivers.

UncleRus, 2024-02-03:
Thank you. But could you do a pull request instead of an issue?

MRusinov, 2024-02-04:
Never did it, but I'll try today...

closed 2024-02-04 by UncleRus