UncleRus/esp-idf-lib #615: Unable to read from HX711 with esp-idf-lib

r3v1, 2024-03-04:
### The issue

I've been using this [HX711](https://github.com/bogde/HX711) Arduino library without any issues, but now I'm moving to this repo, and the readings are always 0 with any change. In the Arduino library the weights are almost perfect. 

I tried to mimic the behaviour of the HX711 component but with no fortune. 

I run the HX711 example, both reading average data and raw data, and no one can.

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

0.9.4

### Which build target have you used?

- [ ] esp32
- [X] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

hx711

### Anything in the logs that might be useful for us?

```console
I (36) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (36) boot: compile time Mar  4 2024 19:17:42
I (37) boot: chip revision: v1.0
I (37) boot.esp32s2: SPI Speed      : 80MHz
I (37) boot.esp32s2: SPI Mode       : DIO
I (38) boot.esp32s2: SPI Flash Size : 2MB
I (38) boot: Enabling RNG early entropy source...
I (38) boot: Partition Table:
I (38) boot: ## Label            Usage          Type ST Offset   Length
I (39) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (40) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (41) boot:  2 factory          factory app      00 00 00010000 00100000
I (43) boot: End of partition table
I (43) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=085a4h ( 34212) map
I (50) esp_image: segment 1: paddr=000185cc vaddr=3ffbeea0 size=0194ch (  6476) load
I (52) esp_image: segment 2: paddr=00019f20 vaddr=40024000 size=060f8h ( 24824) load
I (58) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=135a4h ( 79268) map
I (74) esp_image: segment 4: paddr=000335cc vaddr=4002a0f8 size=04da8h ( 19880) load
I (85) boot: Loaded app from partition at offset 0x10000
I (86) boot: Disabling RNG early entropy source...
I (106) main_task: Started on CPU0
I (106) main_task: Calling app_main()
I (106) main_task: Returned from app_main()
I (116) hx711-example: Raw data: 0
I (616) hx711-example: Raw data: 0
I (1116) hx711-example: Raw data: 0
I (1616) hx711-example: Raw data: 0
I (2116) hx711-example: Raw data: 0
I (2616) hx711-example: Raw data: 0
I (3116) hx711-example: Raw data: 0
I (3616) hx711-example: Raw data: 0
```

### Confirmation

- [X] This report is not a question nor a request for drivers.

r3v1, 2024-03-06:
At the end, it wall all about wrong gpio init. This is the original implementation:

```c
esp_err_t hx711_init(hx711_t *dev)
{
    CHECK_ARG(dev);

    CHECK(gpio_set_direction(dev->dout, GPIO_MODE_INPUT));
    CHECK(gpio_set_direction(dev->pd_sck, GPIO_MODE_OUTPUT));

    CHECK(hx711_power_down(dev, false));

    return hx711_set_gain(dev, dev->gain);
}
```

But this initialitation works better:

```c
esp_err_t hx711_init(hx711_t *dev)
{
    CHECK_ARG(dev);

    // CHECK(gpio_set_direction(dev->dout, GPIO_MODE_INPUT));
    // CHECK(gpio_set_direction(dev->pd_sck, GPIO_MODE_OUTPUT));
    gpio_config_t io_conf;
    io_conf.intr_type = GPIO_INTR_DISABLE;
    io_conf.mode = GPIO_MODE_OUTPUT;
    io_conf.pin_bit_mask = (1ULL<<dev->pd_sck);
    io_conf.pull_down_en = 0;
    io_conf.pull_up_en = 0;
    gpio_config(&io_conf);

    io_conf.intr_type = GPIO_INTR_DISABLE;
    io_conf.pin_bit_mask = (1ULL<<dev->dout);
    io_conf.mode = GPIO_MODE_INPUT;
    io_conf.pull_up_en = 0;
    gpio_config(&io_conf);

    CHECK(hx711_power_down(dev, false));

    return hx711_set_gain(dev, dev->gain);
}
```

Should I pull request?

UncleRus, 2024-03-08:
> Should I pull request?

No, thanks!

PR #617 (2024-03-08): fix: reset gpio before use in hx711
Fixes #615
closed 2024-03-08 by UncleRus