UncleRus/esp-idf-lib #620: Watchdog timer reset after BME280 initialization on ESP32C3

camielverdult, 2024-03-19:
### The issue

When using the code from [the example](https://github.com/UncleRus/esp-idf-lib/blob/master/examples/bmp280/default/main/main.c), I am getting a watchdog timer reset after the bmp280_init function call:

```c
void app_main(void)
{
    ESP_ERROR_CHECK(i2cdev_init());
    // xTaskCreatePinnedToCore(bmp280_test, "bmp280_test", configMINIMAL_STACK_SIZE * 8, NULL, 5, NULL, PRO_CPU_NUM);

    bmp280_params_t params;
    ESP_LOGI("app_main", "a");
    bmp280_init_default_params(&params);
    ESP_LOGI("app_main", "b");
    bmp280_t dev;
    ESP_LOGI("app_main", "c");
    memset(&dev, 0, sizeof(bmp280_t));
    ESP_LOGI("app_main", "d");

    ESP_ERROR_CHECK(bmp280_init_desc(&dev, BMP280_I2C_ADDRESS_1, I2C_NUM_0, I2C_SDA, I2C_SCL));
    ESP_LOGI("app_main", "e");
    ESP_ERROR_CHECK(bmp280_init(&dev, &params));
    ESP_LOGI("app_main", "f");

    bool bme280p = dev.id == BME280_CHIP_ID;
    // Log chip id
    // ESP_LOGI("BMP280: found %s\n", bme280p ? "BME280" : "BMP280");

    ESP_LOGI("app_main", "g");
    float pressure, temperature, humidity;

    while (1)
    {
        ESP_LOGI("app_main", "looping...");
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        if (bmp280_read_float(&dev, &temperature, &pressure, &humidity) != ESP_OK)
        {
            // Log the error
            ESP_LOGE("BMP280", "Temperature/pressure reading failed");
            continue;
        }

        ESP_LOGI("app_main", "Pressure: %.2f Pa, Temperature: %.2f C", pressure, temperature);
        if (bme280p)
            ESP_LOGI("BMP280", ", Humidity: %.2f\n", humidity);
        else
            ESP_LOGI("BMP280", "\n");
    }
}
```

I do not know if this issue is related to #609.

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

master (Wed Aug 23 19:25:22 2023)

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [X] other

### Component causing the issue

bmp280

### Anything in the logs that might be useful for us?

```console
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
riscv32-esp-elf-addr2line -pfiaC -e /Users/camiel/Documents/GitHub/c3-sensor/build/c3-sensor.elf 0x40380082: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1738
load:0x403cc710,len:0xae8
load:0x403ce710,len:0x2e7c
entry 0x403cc71a
I (24) boot: ESP-IDF v5.2-dev-2383-g82cceabc6e 2nd stage bootloader
I (24) boot: compile time Mar 19 2024 09:50:52
I (25) boot: chip revision: v0.4
I (29) boot.esp32c3: SPI Speed      : 80MHz
I (34) boot.esp32c3: SPI Mode       : DIO
I (38) boot.esp32c3: SPI Flash Size : 2MB
I (43) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0a568h ( 42344) map
I (101) esp_image: segment 1: paddr=0001a590 vaddr=3fc8c200 size=01158h (  4440) load
I (104) esp_image: segment 2: paddr=0001b6f0 vaddr=40380000 size=04928h ( 18728) load
I (115) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=19620h (103968) map
I (136) esp_image: segment 4: paddr=00039648 vaddr=40384928 size=07750h ( 30544) load
I (146) boot: Loaded app from partition at offset 0x10000
I (147) boot: Disabling RNG early entropy source...
I (158) cpu_start: Unicore app
I (158) cpu_start: Pro cpu up.
I (167) cpu_start: Pro cpu start user code
I (167) cpu_start: cpu freq: 160000000 Hz
I (167) cpu_start: Application information:
I (170) cpu_start: Project name:     c3-sensor
I (175) cpu_start: App version:      88b924b-dirty
I (180) cpu_start: Compile time:     Mar 19 2024 22:31:39
I (186) cpu_start: ELF file SHA256:  d393f7b95...
I (192) cpu_start: ESP-IDF:          v5.2-dev-2383-g82cceabc6e
I (198) cpu_start: Min chip rev:     v0.3
I (203) cpu_start: Max chip rev:     v0.99 
I (208) cpu_start: Chip rev:         v0.4
I (213) heap_init: Initializing. RAM available for dynamic allocation:
I (220) heap_init: At 3FC8E200 len 0004E510 (313 KiB): DRAM
I (226) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (233) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (240) spi_flash: detected chip: generic
I (244) spi_flash: flash io: dio
W (247) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
W (261) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (271) sleep: Configure to isolate all GPIO pins in sleep state
I (278) sleep: Enable automatic switching of GPIO sleep configuration
I (285) app_start: Starting scheduler on CPU0
I (290) main_task: Started on CPU0
I (290) main_task: Calling app_main()
I (290) app_main: 1
I (300) app_main: a
I (300) app_main: b
I (300) app_main: c
I (300) app_main: d
I (310) app_main: e
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380080
riscv32-esp-elf-addr2line -pfiaC -e /Users/camiel/Documents/GitHub/c3-sensor/build/c3-sensor.elf 0x40380080: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1738
load:0x403cc710,len:0xae8
load:0x403ce710,len:0x2e7c
entry 0x403cc71a
I (24) boot: ESP-IDF v5.2-dev-2383-g82cceabc6e 2nd stage bootloader
I (24) boot: compile time Mar 19 2024 09:50:52
I (25) boot: chip revision: v0.4
I (28) boot.esp32c3: SPI Speed      : 80MHz
I (33) boot.esp32c3: SPI Mode       : DIO
I (38) boot.esp32c3: SPI Flash Size : 2MB
W (43) boot.esp32c3: PRO CPU has been reset by WDT.
... omitted duplicate logs ...
I (295) main_task: Calling app_main()
I (295) app_main: 1
I (305) app_main: a
I (305) app_main: b
I (305) app_main: c
I (305) app_main: d
I (315) app_main: e
```

### Additional information or context

This is with the Adafruit BME280 breakout board on a generic ESP32-C3-MINI-1 dev board.

### Confirmation

- [X] This report is not a question nor a request for drivers.

camielverdult, 2024-03-19:
The [I2C scanner](https://github.com/UncleRus/esp-idf-lib/blob/master/examples/i2cdev/default/main/main.c) also throws this watchdog timer reset. 

```
I (294) main_task: Calling app_main()
I (294) app_main: 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
riscv32-esp-elf-addr2line -pfiaC -e /Users/camiel/Documents/GitHub/c3-sensor/build/c3-sensor.elf 0x40380082: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1738
load:0x403cc710,len:0xae8
load:0x403ce710,len:0x2e7c
entry 0x403cc71a
I (24) boot: ESP-IDF v5.2-dev-2383-g82cceabc6e 2nd stage bootloader
I (24) boot: compile time Mar 19 2024 09:50:52
I (25) boot: chip revision: v0.4
I (28) boot.esp32c3: SPI Speed      : 80MHz
I (33) boot.esp32c3: SPI Mode       : DIO
I (38) boot.esp32c3: SPI Flash Size : 2MB
W (43) boot.esp32c3: PRO CPU has been reset by WDT.
```

lszpsf0, 2025-01-17:
I had the same problem

lszpsf0, 2025-01-17:
I (35) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (35) boot: compile time Jan 17 2025 10:53:29
I (35) boot: chip revision: v0.4
I (39) boot.esp32c3: SPI Speed      : 80MHz
I (43) boot.esp32c3: SPI Mode       : DIO
I (48) boot.esp32c3: SPI Flash Size : 4MB
W (53) boot.esp32c3: PRO CPU has been reset by WDT.
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (82) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00100000
I (97) boot: End of partition table
I (101) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0a0a0h ( 41120) map
I (116) esp_image: segment 1: paddr=0001a0c8 vaddr=3fc8cc00 size=01394h (  5012) load
I (119) esp_image: segment 2: paddr=0001b464 vaddr=40380000 size=04bb4h ( 19380) load
I (130) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=18d3ch (101692) map
I (151) esp_image: segment 4: paddr=00038d64 vaddr=40384bb4 size=07ff4h ( 32756) load
I (161) boot: Loaded app from partition at offset 0x10000
I (161) boot: Disabling RNG early entropy source...
I (173) cpu_start: Unicore app
I (181) cpu_start: Pro cpu start user code
I (181) cpu_start: cpu freq: 160000000 Hz
I (182) app_init: Application information:
I (184) app_init: Project name:     5611-sp
I (189) app_init: App version:      1
I (194) app_init: Compile time:     Jan 17 2025 10:53:16
I (200) app_init: ELF file SHA256:  c87dd98fe...
I (205) app_init: ESP-IDF:          v5.3.1-dirty
I (210) efuse_init: Min chip rev:     v0.3
I (215) efuse_init: Max chip rev:     v1.99 
I (220) efuse_init: Chip rev:         v0.4
I (225) heap_init: Initializing. RAM available for dynamic allocation:
I (232) heap_init: At 3FC8EE80 len 00031180 (196 KiB): RAM
I (238) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (245) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (252) heap_init: At 50000200 len 00001DE8 (7 KiB): RTCRAM
I (259) spi_flash: detected chip: generic
I (263) spi_flash: flash io: dio
W (267) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (278) sleep: Configure to isolate all GPIO pins in sleep state
I (284) sleep: Enable automatic switching of GPIO sleep configuration
I (292) main_task: Started on CPU0
I (302) main_task: Calling app_main()

trombik, 2025-05-23:
To narrow down the cause, would you do the following?

* do `git bisect` on _ESP-IDF_ (NOT our code) with our code at a particular commit.
* if the above does not identify a commit ref that reproduces the issue (i.e. all the tests fail), do `git bisect` on _our code_ but using the master branch of _ESP-IDF_.

You can read a tutorial of `git bisect` at: https://www.geeksforgeeks.org/git-bisect/

trombik, 2025-05-23:
depends on #701