UncleRus/esp-idf-lib #624: [sht3x]: i2cdev: Could not write to device [0x44 at 0]

FeOAr, 2024-03-30:
### The issue

- release version: 0.9.4
- esp-idf version: 5.1.1

[image]

I am using the example of SHT30, which can function normally when the I2C speed is 100K, but other chips can reach 200K using the same "i2cdev" lib. The last discovered issue may be that the i2c device was not fully configured before the first measurement. Just add a delay to use it normally.

```c++
void task(void *pvParameters)
{
    float temperature;
    float humidity;

    TickType_t last_wakeup = xTaskGetTickCount();
    vTaskDelay(pdMS_TO_TICKS(10));  // Add delay before measurement

    while (1)
    {
        // perform one measurement and do something with the results
        ESP_ERROR_CHECK(sht3x_measure(&dev, &temperature, &humidity));
        printf("SHT3x Sensor: %.2f °C, %.2f %%\n", temperature, humidity);

        // wait until 5 seconds are over
        vTaskDelayUntil(&last_wakeup, pdMS_TO_TICKS(1000));
    }
}
```

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

5.1.1

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [X] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

sht3x

### Anything in the logs that might be useful for us?

```console
E (320) i2cdev: Could not write to device [0x44 at 0]: -1 (ESP_FAIL)
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4200870b
```

### Confirmation

- [X] This report is not a question nor a request for drivers.

cmorganBE, 2024-05-10:
Using sht4x driver here, 400khz. Seeing similar issues with period failures during configuration (retries resolve), and periodically during run-time where the write to start a measurement read fails:

```
E (668146) i2cdev: Could not read from device [0x44 at 0]: -1 (ESP_FAIL)
```

Note that these failures are seemingly random, we read the temp/humidity every 500ms, several reads are successful, then a failure, then the next several are successful etc.

cmorganBE, 2024-05-10:
sht4x is rated for 1MHz i2c, as is sht3x. Signals on the scope look clean here. Haven't tried slowing the i2c comms rate yet.

cmorganBE, 2024-05-10:
And testing here with 100khz doesn't resolve the issue here...

cmorganBE, 2024-05-13:
I was able to catch the issue here. In the good case:

- write to start measurement (0xFD)
- Longer delay (~1ms)
- data response from part

bad case:
- write to start measurement (0xFD)
- Shorter delay (~0.5ms)
- Data response is cut short after the first byte

cmorganBE, 2024-05-14:
Alright, root cause identified for my case, opening a PR today to resolve. The issue was that for the SH4Tx driver pdMS_TO_TICKS() was being used. ticks on my system (and likely many others) is 10ms (100hz). The delay time was 10ms / (10ms / tick) = 1 tick, but vTaskDelay() is NOT guaranteed to delay for at least the value given, see https://www.freertos.org/a00127.html, there is no mention of "at least" which is typical for delay functions. So for single tick delays you *could* delay for 10ms, or 6ms or 15ms, it does its best to get close.

In the case where the delay is supposed to be 10ms but its 6ms the SH4Tx isn't ready yet to respond, hence the error.

This is corroborated by:
- Scope showing 10ms (1 tick) delays being shorter than 10ms (otherwise I wouldn't have believed it)
- SH4Tx measurements only fail if the time between the 0xFD (start measurement) and the measurement read is less than 10ms
- No failures if I apply the fixes from the SHT3x driver to the SHT4x driver (this is going to be the PR I'm opening today)

So maybe this isn't the same issue as the OP reported with the sht3x not reading correctly since it looks like that driver has had the correct tick calculation with margin in place since 8dd5853b (from 2019).

FeOAr, 2024-05-14:
Thank you very much for your answer. I now have a rough understanding. I will take some time to reproduce it with an oscilloscope later.

cmorganBE, 2024-05-14:
@FeOAr I'm guessing you have a different issue than I have. I wouldn't have posted here but it looked like the same issue initially. PR is #632

trianglesis, 2025-04-19:
Hello,
I'm seeing a similar issue with SCD41.

```log
I (10157) CO2 SCD41 INIT: run scd4x_get_serial_number
D (10167) i2cdev: Reconfiguring I2C driver on port 0
D (10167) intr_alloc: Connected src 50 to int 15 (cpu 0)
D (10177) i2cdev: I2C driver successfully reconfigured on port 0
D (10187) i2cdev: Timeout: ticks = 0 (0 usec) on port 0
D (10177) wifi:mms: 0->0
E (10187) i2cdev: Could not write to device [0x62 at 0]: -1 (ESP_FAIL)
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x42011308
--- 0x42011308: sensor_init at D:/Projects/ESP/projects/ESP32-C6-OLED/Air_Quality_station/main/sensor_co2/co2_sensor.c:164 (discriminator 1)

file: "./main/sensor_co2/co2_sensor.c" line 164
func: sensor_init
expression: scd4x_get_serial_number(&dev, serial, serial + 1, serial + 2)

abort() was called at PC 0x4080b6d1 on core 0
--- 0x4080b6d1: _esp_error_check_failed at D:/Projects/ESP/Espressif/v5.4.1/esp-idf/components/esp_system/esp_err.c:49
```

I've read datasheets for this sensor and tried to switch two of them, one working on an ESPHome device.

I have a complete run here, there are a lot of words...

[README.md](https://github.com/trianglesis/Air_Quality_station/blob/4802c416501cf49720c3299119afd54bb335f894/doc/how_to_basics/README.md#scd4x)

trombik, 2025-08-01:
related: https://github.com/esp-idf-lib/i2cdev/issues/4 https://github.com/esp-idf-lib/scd4x/issues/2

trombik, 2025-08-01:
moved to: https://github.com/esp-idf-lib/sht3x/issues/1

closed 2025-08-01 by trombik