UncleRus/esp-idf-lib #626: I2C driver is old driver in i2cdev

FilipVBrugghe, 2024-05-02:
### The issue

When using the driver i2cdev I got the message "i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`"

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

master

### Which build target have you used?

- [X] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

ds3231 with i2cdev

### Anything in the logs that might be useful for us?

```console
The component i2cdev is using the old driver i2c.h which has a newer version i2c_master.h
Will the component i2cdev updated?
```

### Confirmation

- [X] This report is not a question nor a request for drivers.

PaulskPt, 2024-05-14:
I encounter the same issue using an AHT20 with aht and i2cdev

UncleRus, 2024-05-16:
1. This is just a message warning you that you are using an old version of the driver. It does not affect the operation of the library and components in any way.
2. I am currently working on a painless transition to a new driver version while maintaining compatibility with all existing code.

closed 2024-05-16 by UncleRus