UncleRus/esp-idf-lib #628: mcp23x17 Interrupt Mirroring

KJ7NYE, 2024-05-03:
### The issue

Add function to library to enable mirroring of interrupt pins (Bit 6 of IOCON) Register

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

master

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [X] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

MCP23x17

### Confirmation

- [X] This report is not a question nor a request for drivers.

KJ7NYE, 2024-05-03:
The contribution guidelines say to assign to self if working on I am unsure how to do this.

trombik, 2025-08-01:
moved to: https://github.com/esp-idf-lib/mcp23x17/issues/1

closed 2025-08-01 by trombik