UncleRus/esp-idf-lib #666: sht3x : E (1307) i2cdev: Could not write to device [0x44 at 0]: 263 (ESP_ERR_TIMEOUT)

mehmetbozhaydar, 2024-11-11:
### The issue

I am getting an error while running the sht3x example with the esp32-h2 module.

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

5.1.0

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [X] other

### Component causing the issue

SHT3x

### Anything in the logs that might be useful for us?

```console
SPIWP:0xee
mode:DIO, clock div:1
load:0x4083cfd0,len:0xc10
load:0x4083efd0,len:0x2d2c
load:0x408460e0,len:0x1778
entry 0x4083cfd0
I (27) boot: ESP-IDF v5.1 2nd stage bootloader
I (27) boot: compile time Nov 11 2024 12:09:20
I (29) boot: chip revision: v0.1
I (30) boot.esp32h2: SPI Speed      : 64MHz
I (34) boot.esp32h2: SPI Mode       : DIO
I (39) boot.esp32h2: SPI Flash Size : 2MB
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=42018020 size=08db8h ( 36280) map
I (112) esp_image: segment 1: paddr=00018de0 vaddr=40800000 size=07238h ( 29240) load
I (128) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=16264h ( 90724) map
I (170) esp_image: segment 3: paddr=0003628c vaddr=40807238 size=02f70h ( 12144) load
I (178) esp_image: segment 4: paddr=00039204 vaddr=4080a1b0 size=00e04h (  3588) load
I (188) boot: Loaded app from partition at offset 0x10000
I (189) boot: Disabling RNG early entropy source...
I (199) cpu_start: Unicore app
I (200) cpu_start: Pro cpu up.
W (209) clk: esp_perip_clk_init() has not been implemented yet
I (216) cpu_start: Pro cpu start user code
I (216) cpu_start: cpu freq: 96000000 Hz
I (217) cpu_start: Application information:
I (219) cpu_start: Project name:     example-sht3x
I (224) cpu_start: App version:      1
I (229) cpu_start: Compile time:     Nov 11 2024 12:09:03
I (235) cpu_start: ELF file SHA256:  3c2a2d18bb5c059a...
I (241) cpu_start: ESP-IDF:          v5.1
I (246) cpu_start: Min chip rev:     v0.0
I (250) cpu_start: Max chip rev:     v0.99 
I (255) cpu_start: Chip rev:         v0.1
I (260) heap_init: Initializing. RAM available for dynamic allocation:
I (267) heap_init: At 4080BEB0 len 000414D0 (261 KiB): D/IRAM
I (273) heap_init: At 4084D380 len 00002B60 (10 KiB): STACK/DIRAM
I (281) spi_flash: detected chip: generic
I (285) spi_flash: flash io: dio
W (289) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (302) app_start: Starting scheduler on CPU0
I (307) main_task: Started on CPU0
I (307) main_task: Calling app_main()
E (1307) i2cdev: Could not write to device [0x44 at 0]: 263 (ESP_ERR_TIMEOUT)
ESP_ERROR_CHECK failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x42005d44
--- 0x42005d44: app_main at C:/Users/Haydar/Videos/sdsf/main/main.c:131 (discriminator 1)

file: "./main/main.c" line 131
func: app_main
expression: sht3x_init(&dev)

abort() was called at PC 0x40804fa3 on core 0
--- 0x40804fa3: _esp_error_check_failed at C:/Users/Haydar/esp/v5.1/esp-idf/components/esp_system/esp_err.c:50

Core  0 register dump:
MEPC    : 0x408004fa  RA      : 0x40804fae  SP      : 0x4080ec10  GP      : 0x4080a9b0
--- 0x408004fa: panic_abort at C:/Users/Haydar/esp/v5.1/esp-idf/components/esp_system/panic.c:452
0x40804fae: __ubsan_include at C:/Users/Haydar/esp/v5.1/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x40806048  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
--- 0x40806048: prvInitialiseNewTask at C:/Users/Haydar/esp/v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:1171 (discriminator 3)

S0/FP   : 0x00000004  S1      : 0x4080ec74  A0      : 0x4080ec3c  A1      : 0x4080ec72
A2      : 0x00000000  A3      : 0x4080ec69  A4      : 0x00000001  A5      : 0x4080c000
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
--- 0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4080ec10: 0x4201ae48 0x42005d48 0x4080ec70 0x408090f4 0x00000000 0x00000000 0x4084db58 0x4080ac14
--- 0x42005d48: cfg_equal at C:/Users/Haydar/Desktop/components/i2cdev/i2cdev.c:178
0x408090f4: abort at C:/Users/Haydar/esp/v5.1/esp-idf/components/newlib/abort.c:35 (discriminator 3)

4080ec30: 0x4080ec74 0x4080ac30 0x4080ec70 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
4080ec50: 0x43502074 0x34783020 0x34303830 0x20336166 0x63206e6f 0x2065726f 0x00000030 0x00000000
4080ec70: 0x00000030 0x30383034 0x33616634 0x00000000 0x00000000 0x00000000 0x4080c000 0x40804fa6
--- 0x40804fa6: esp_system_abort at C:/Users/Haydar/esp/v5.1/esp-idf/components/esp_system/port/esp_system_chip.c:83

4080ec90: 0x00000000 0x4084da60 0x4201a000 0x42005d48 0x4201a428 0x00000133 0x4201a000 0x42015e8a
--- 0x42005d48: cfg_equal at C:/Users/Haydar/Desktop/components/i2cdev/i2cdev.c:178
0x42015e8a: main_task at C:/Users/Haydar/esp/v5.1/esp-idf/components/freertos/app_startup.c:209 (discriminator 13)

4080ecb0: 0x00000000 0x00001388 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x408074a2
--- 0x408074a2: vPortTaskWrapper at C:/Users/Haydar/esp/v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205

4080ecd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4080ecf0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x4080ead0 0x00000064
4080ed10: 0x4080b170 0x4080b170 0x4080ed08 0x4080b168 0x00000018 0x4084dca4 0x4084dca4 0x4080ed08
4080ed30: 0x00000000 0x00000001 0x4080dd04 0x6e69616d 0x308d0800 0x546005c4 0x00fe6cef 0x00000000
4080ed50: 0x4080ed00 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x4080c864 0x4080c8cc
4080ed70: 0x4080c934 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x40011484
--- 0x40011484: _cleanup_r in ROM

4080ed90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4080edb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4080edd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4080edf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4080ee10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4080ee30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4080ee50: 0x00000000 0x00000000 0x40000000 0x00000600 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
--- 0x40000000: _start in ROM

4080ee70: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4080ee90: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4080eeb0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4080eed0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4080eef0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4080ef10: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4080ef30: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4080ef50: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4080ef70: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4080ef90: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4080efb0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4080efd0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4080eff0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 3c2a2d18bb5c059a

Rebooting...
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400031b6
--- 0x400031b6: software_reset_cpu in ROM
```

### Confirmation

- [X] This report is not a question nor a request for drivers.

trombik, 2024-11-11:
Please read the FAQ and make sure the I2C device works and no wiring issues is in your circuit.

mehmetbozhaydar, 2024-11-11:
Thanks, the sensor is burned out.

closed 2024-11-11 by mehmetbozhaydar