UncleRus/esp-idf-lib #667: Add support for new I2C driver used in IDF 5.4

elijahnyp, 2024-11-17:
### The issue

with the upcoming release of 5.4 (currently in beta), there is a new I2C driver.  The new driver will not co-exist with the old driver (i2c.h vs i2c_master.h).  Some libraries, like the esp32-camera library, support and use the new driver when compiled with version 5.4 of the IDF.  Without updating to support the new driver, the esp-idf-lib drivers are incapable of co-existing with drivers that do use the new setup.

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

v5.4

### Which build target have you used?

- [X] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp8266
- [ ] other

### Component causing the issue

i2cdev

### Additional information or context

The issue occurs at runtime due to changes in i2c.c in idf 5.4.  If the new driver is linked in, at runtime the old i2c driver aborts execution.
```
/**
 * @brief This function will be called during start up, to check that this legacy i2c driver is not running along with the new I2C driver
 */
__attribute__((constructor))
static void check_i2c_driver_conflict(void)
{
    // This function was declared as weak here. The new I2C driver has the implementation.
    // So if the new I2C driver is not linked in, then `i2c_acquire_bus_handle()` should be NULL at runtime.
    extern __attribute__((weak)) esp_err_t i2c_acquire_bus_handle(int port_num, void *i2c_new_bus, int mode);
    if ((void *)i2c_acquire_bus_handle != NULL) {
        ESP_EARLY_LOGE(I2C_TAG, "CONFLICT! driver_ng is not allowed to be used with this old driver");
        abort();
    }
    ESP_EARLY_LOGW(I2C_TAG, "This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`");
}
```

### Confirmation

- [X] This report is not a question nor a request for drivers.

AxelLin, 2024-12-11:
Hi @UncleRus 
Is there any update for transition to a new driver version? https://github.com/UncleRus/esp-idf-lib/issues/626#issuecomment-2114567234

AxelLin, 2025-01-07:
Hi @UncleRus 
Any update? FYI, the IDF-5.4 is already released.

Graziele-Rodrigues, 2025-01-30:
same issue here

AxelLin, 2025-05-06:
Hi @trombik 
Any update for this issue?

commit 8faefbe2 (2025-07-10): feat: migrate i2cdev to new ESP-IDF I2C master driver API

- Replace legacy command-based I2C with handle-based master API
- Add device handle tracking and automatic resource cleanup
- Enhance addressing support (7-bit/10-bit) and error handling
- Implement smart retry logic with exponential backoff
- Add buffer overflow protection and memory optimization
- Maintain backward compatibility via i2cdev_legacy.c bridge
- Preserve existing sensor driver API compatibility
potentialy fixes #667 #713 #741