UncleRus/esp-idf-lib #674: The project status

trombik, 2025-02-04:
### The issue

The project seems to halt. The project is lagged behind the latest esp-idf version. The I2C-based drivers cannot be used with third-party I2C drivers because of the conflict with the latest esp-idf I2C driver.

### The context

If you are wondering what happened to the project, the latest IDF support, or ESP8266 support, please see #664 for the explanation. 

### Possible solution

See the PR.

### Confirmation

- [x] This report is not a bug report in code, a question, nor a request for drivers.

jmpmscorp, 2025-02-25:
Hi @trombik.

Any guidelines to port to i2c_master?

I have been porting i2c lib to i2c master trying to do minimal changes. I get working code (need more exhaustive test) with same write/read api. I have tested ads111x and ds3231 examples with a real device and they work.

However, there is a challenge to solve with ```i2c_dev_t``` structure. On every i2c device constructor, there is an init descriptor function like ```xxxx_init_desc(...)```. In .c code, descriptor is filled. With i2c drive legacy, ```i2c_dev_t``` has a member (```i2c_config_t cfg```) but in i2c_master that struct doesn't exist.

With i2c_master, ```i2c_dev_t``` only need ```i2c_master_dev_handle_t``` and optionally, ```i2c_master_bus_handle". The definition with new driver could be:

```
typedef struct
{
    uint8_t addr;    //!< Unshifted address
    i2c_port_t port; //!< I2C port number
    uint32_t timeout_ticks;  //!< Threshold value (stretch time)

#if USE_I2C_LEGACY_DRIVER
    i2c_config_t cfg;       //!< I2C driver configuration
#else  // USE I2C MASTER DRIVER
    i2c_master_dev_handle_t dev_handle; //!< I2C driver handle
    i2c_master_bus_handle_t bus_handle; //!< Optional I2C bus handle
#endif
    SemaphoreHandle_t mutex; //!< Device mutex

} i2c_dev_t;
```

This struct change implies that every i2c component must rewrite its initializer with the new structure, something like: 
```
xxx_init_desc(...)
{
  #if USE_I2C_LEGACY
       initliaze i2c legacy structure
  #else
      initialize i2c master structure
  #endif
}
```

An alternative could be create new global "init/constructor" in i2cdev.h with a new global config structure. In this way, components only need to call common initializer and descriptor/device initialization will be done in i2c_dev_t component. Something like:

i2c_dev.h

```
typedef struct
{
    i2c_port_t port;
    uint16_t address;
    i2c_clock_source_t clk_src;
    uint8_t sda_gpio_num;
    uint8_t scl_gpio_num;
    uint32_t scl_speed_hz;
    bool enable_gpio_pullups;
    bool allow_pd;
    uint32_t timeout_ticks;
} i2c_dev_config_t;

esp_err_t i2c_dev_new(i2c_dev_t *dev, const i2c_dev_config_t *config);

```

i2c_legacy.c

```
esp_err_t i2c_dev_new(i2c_dev_t *dev, const i2c_dev_config_t *config)
{
    required work with i2c legacy driver
}
```

```
esp_err_t i2c_dev_new(i2c_dev_t *dev, const i2c_dev_config_t *config)
{
    required work with i2c master driver
}
```

xxxx_component.c

```
esp_err_t xxx_init_desc(....)
{
   // fill config structure
   
   return i2c_dev_new(...)
}
```

Obviously, every i2c component must rewrite its initializer to accomodate to this change.

I'm open to any suggestion.

Regards.

trombik, 2025-03-01:
@jmpmscorp, would you create a discussion thread? Just copying and pasting your comment above are enough. I would like focus on the project status in this issue. Thanks.

jmpmscorp, 2025-03-01:
Just move to a discussion #688.

trombik, 2025-08-01:
Closing because https://github.com/esp-idf-lib/ is the new home.