UncleRus/esp-idf-lib #698: DHT22 with esp32 s3 gives phase b error

anjula544, 2025-05-13:
### The issue

Sensor give the readings but the have a issue with phase B

I (60252) sensor_dht: -- Reading DHT Sensor values
E (60277) dht: Initialization error, problem in phase 'B'
W (60278) sensor_dht: Retrying DHT read (1/3)
I (61302) sensor_dht: -- Reading DHT Sensor values
E (61322) dht: Initialization error, problem in phase 'B'
W (61323) sensor_dht: Retrying DHT read (1/3)
I (62273) JSON_settings: -- Dot Config:

used library-zorexx_dht

wirings,supply voltage and pullup resistor alredy applied correctly.

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

5.1.3

### Which build target have you used?

- [ ] esp32
- [ ] esp32s2
- [x] esp32s3
- [ ] esp32c2
- [ ] esp32c5
- [ ] esp32c6
- [ ] esp32c61
- [ ] esp32h2
- [ ] esp32p4
- [ ] esp8266
- [ ] other

### Component causing the issue

dht22

### Anything in the logs that might be useful for us?

```console

```

### Confirmation

- [x] This report is not a question nor a request for drivers.

Ksisu, 2025-05-14:
I had this error before. My working solution:
- power up the sensor
- wait 2 sec (from sensor documentation "When power is supplied to sensor, don't send any instruction to the sensor
within one second to pass unstable status.")
- read the sensor (it will fail most of the time; idk why)
- wait 100ms
- read the sensor again

trombik, 2025-05-23:
@Ksisu Thanks for a possible workaround (or a solution). 

@anjula544 I don't have the device and cannot reproduce the issue. Would you try the workaround suggested by @Ksisu? 

* Write a simple proof of concept to reproduce the issue.
* Copy the proof of concept and modified it as suggested by @Ksisu 
* Run the both programs with an identical circuit
* Report the results of the both cases here

Dumbo-1007, 2025-06-02:
```C
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "OLED.h"
#include "mydht11.h"
#include "dht.h"
#include "esp_log.h" // 确保包含日志头文件

#define OLED_I2C I2C_NUM_0
#define OLED_SCL 4
#define OLED_SDA 3
#define OLED_ADD 0x78
#define OLED_SPEED 400000
#define DHT_SENSOR_PIN 18
#define DHT_SENSOR_TYPE DHT_TYPE_DHT11
#define DELAY_TIME 3000

static const char *TAG = "OLED_DHT";
void display_temp_hum(float temperature, float humidity)
{
  OLED_Clear();

  OLED_ShowString(0, 0, "hum:", OLED_8X16);
  OLED_ShowString(0, 16, "tmp:", OLED_8X16);

  OLED_ShowFloatNum(32, 0, humidity,
                    2, 2, OLED_8X16);
  OLED_ShowFloatNum(32, 16, temperature,
                    2, 2, OLED_8X16);
  OLED_Update();
}

void dht_task(void *pvParameter)
{
  float temperature = 0.0;
  float humidity = 0.0;

  while (1)
  {
    esp_err_t result = dht_read_float_data(DHT_SENSOR_TYPE,DHT_SENSOR_PIN,&humidity, &temperature);
    if (result == ESP_OK)
    {
      ESP_LOGI(TAG, "Temperature: %.1f C, Humidity: %.1f %%", temperature, humidity);
      display_temp_hum(temperature, humidity);
    }
    else
    {
      ESP_LOGE(TAG, "Failed to read data from DHT sensor: %s", esp_err_to_name(result));
      OLED_Clear();
      OLED_ShowString(0, 0, "Sensor Error", OLED_8X16);
      OLED_Update();
    }
    vTaskDelay(pdMS_TO_TICKS(2000));
  }
}

void app_main(void)
{
  OLED_Init(OLED_I2C, OLED_ADD, OLED_SCL, OLED_SDA, OLED_SPEED);
  gpio_set_pull_mode(DHT_SENSOR_PIN, GPIO_PULLUP_ONLY);
  xTaskCreate(dht_task, "DHT_OLED_Task", 2048, NULL, 5, NULL);
}
```

error:
E (12751) OLED_DHT: Failed to read data from DHT sensor: ESP_ERR_TIMEOUT
E (14791) dht: Initialization error, problem in phase 'B'
E (14791) dht: Initialization error, problem in phase 'B'
E (14791) OLED_DHT: Failed to read data from DHT sensor: ESP_ERR_TIMEOUT
E (14791) OLED_DHT: Failed to read data from DHT sensor: ESP_ERR_TIMEOUT
E (16831) dht: Initialization error, problem in phase 'B'
E (16831) OLED_DHT: Failed to read data from DHT sensor: ESP_ERR_TIMEOUT
E (16831) OLED_DHT: Failed to read data from DHT sensor: ESP_ERR_TIMEOUT
E (18871) dht: Initialization error, problem in phase 'B'

trombik, 2025-08-01:
moved to https://github.com/esp-idf-lib/dht/issues/1

closed 2025-08-01 by trombik