UncleRus/esp-idf-lib #699: ultrasonic measure with temperature compensation returns roundtrip

Ksisu, 2025-05-14:
### The issue

I think `ultrasonic_measure_cm_temp_compensated` and `ultrasonic_measure_temp_compensated` return not the distance but the double of the distance (measuring the roundtrip).

example code:
```c
    float temperature_c = 23.0f;
    float distance_m, distance_t_m;
    uint32_t distance_cm, distance_t_cm;
    float max_distance_m = 2.0f;
    uint32_t max_distance_cm = 200;

    ultrasonic_measure(&sensor, max_distance_m, &distance_m);
    sleep(1);
    ultrasonic_measure_temp_compensated(&sensor, max_distance_m, &distance_t_m, temperature_c);
    sleep(1);
    ultrasonic_measure_cm(&sensor, max_distance_cm, &distance_cm);
    sleep(1);
    ultrasonic_measure_cm_temp_compensated(&sensor, max_distance_cm, &distance_t_cm, temperature_c);

    printf("distance_m   : %.4f\n", distance_m);
    printf("distance_t_m : %.4f\n", distance_t_m);
    printf("distance_cm  : %lu\n", distance_cm);
    printf("distance_t_cm: %lu\n", distance_t_cm);
```

with ~11cm:
```
distance_m   : 0.1162
distance_t_m : 0.2327
distance_cm  : 11
distance_t_cm: 23
```

with ~22cm:
```
distance_m   : 0.2188
distance_t_m : 0.4374
distance_cm  : 21
distance_t_cm: 43
```

with ~38cm
```
distance_m   : 0.3810
distance_t_m : 0.7632
distance_cm  : 38
distance_t_cm: 76
```

### Which SDK are you using?

esp-idf

### Which version of SDK are you using?

master

### Which build target have you used?

- [x] esp32
- [ ] esp32s2
- [ ] esp32s3
- [ ] esp32c2
- [ ] esp32c5
- [ ] esp32c6
- [ ] esp32c61
- [ ] esp32h2
- [ ] esp32p4
- [ ] esp8266
- [ ] other

### Component causing the issue

ultrasonic

### Anything in the logs that might be useful for us?

```console

```

### Confirmation

- [x] This report is not a question nor a request for drivers.

Ksisu, 2025-05-14:
A quickfix idea: rename the const to `HALF_THE_SPEED_OF_SOUND_AT_0C_M_S` and use `165.7` (`=331.4/2`)

https://github.com/UncleRus/esp-idf-lib/blob/a02cd6bb5190cab379125140780adcb8d88f9650/components/ultrasonic/ultrasonic.c#L51

If it is okey to you i am happy to create a PR.

trombik, 2025-05-23:
@Ksisu Thanks for reporting an issue and the analysis. Would you create a PR? I think I have the sensor in my stock bin.

closed 2025-07-24 by trombik